stm32duino/Arduino_Core_STM32 #1: analogWrite() on low numbered digital pins not working

madman2003, 2017-05-14:
I have a Nucleo64 L476RG, and i noticed i cannot do PWM using the analogWrite() function when using D2, while i can with D8. D2 being wired to PA10 should have this capability.

I found this in variant.h which looks suspicious:
// Convert an analog pin number Axx to a PinName Pxy
#define analogToPinName(p) ((p < NUM_ANALOG_INPUTS) ? digitalToPinName(p+A0) : digitalToPinName(p))

This function is then used by analogWrite()

fpistm, 2017-05-15:
Hi madaman,
it's a known restriction.
In fact, analogRead/write take as argument the analog pin number. It could be 0 to 5 or A0 to A5.
As 0 is the same pin number as D0 that's why when you use the D0 to D5 it was convert to A0 to A5.
For digitalRead/Write 0 is D0 and A0 -A5 could be used also.
As a workaround, if you want to use D0 to D5, it is possible to duplicate the D0 - D5 in variant.cpp at the end of the  const PinName digital_arduino[]
```
  PA0,  //D46/A0
  PA1,  //D47/A1
  PA4,  //D48/A2
  PB0,  //D49/A3
  PC1,  //D50/A4
  PC0,  //D51/A5
// Duplicate entry to be able to use D0-D5 with analogRead/Write
  PA3,  //D52 = D0
  PA2,  //D53 = D1
  PA10, //D54 = D2
  PB3,  //D55 = D3
  PB5,  //D56 = D4
  PB4  //D57 = D5
```

Don't forget to extend the enum in variant .h to match the digital pin number:
```
  D50, D51, D52, D53, D54, D55, D56, D57,
  DEND
```
I have tried to find a better solution to not duplicate it but I didn't find it.

madman2003, 2017-05-15:
Analog pins are supposed to be referred to as Ax always from what I know? Where does it say that you can drop the A prefix?

fpistm, 2017-05-16:
Here for example:
[](https://www.arduino.cc/en/Reference/AnalogRead)
_pin: the number of the analog input pin to read from (0 to 5 on most boards, 0 to 7 on the Mini and Nano, 0 to 15 on the Mega)_ 
Means A0 to A5, A0 to A7 or A0 to A15.
In the example:
int analogPin = 3;     // potentiometer wiper (middle terminal) connected to analog pin 3

Moreover on several lib, pin number (btw 0 to 5)  is used instead of alias A0 to A5.
This is a  subject which has been discussed several time. 

It could be the solution to force using alias Ax for analog. And if 0 is used it point to D0 (risk this could break some basic examples).
It will be more "restrictive" but more logical.

What do you think about that?

rogerclarkmelbourne, 2017-05-16:
@madman2003 

IMHO its not ever going to be possible to strictly replicate the whole Arduino API, which was developed for 8 bit AVR hardware, to completely different MCU's, like the STM32.

In terms of this restriction, AFIK it is mainly because the method that the Core uses to map Arduino API style pin numbers, to the physical Port & Pin.

On the F1 core, an alternative approach was evolved,  e.g.

https://github.com/stm32duino/Arduino_Core_STM32F1/commit/8fba3e88178e57c30de7622fb149fbfe397ca7b9

and

https://github.com/stm32duino/Arduino_Core_STM32F1/blob/44d46ebd89d0da52c62f268f66821228ae7edc76/variants/STM32F103RB_Nucleo/variant.h#L169-L174

But I'm not sure if this approach would resolve your issue

fpistm, 2017-05-16:
I will correct it like that: 0 will be D0 for analog api and not A0.

closed 2017-05-17 by fpistm

madman2003, 2017-05-17:
An alternative would have been to always treat A0 as analog, D0 as digital, and without prefix depending on the API being used. My problem was that using D0 somehow aliased into the same number as 0, and thus ending up being analog.

commit 73ab64c1 (2017-05-17): Fix #1

For analog api 0 is now D0.
commit b87f9e56 (2017-05-22): Fix #1

For analog api 0 is now D0.