stm32duino/Arduino_Core_STM32 #3: Uploader tool wrong path?

fprwi6labs, 2017-05-24:
I think I missed some thing. Why change the path in platform.txt `{runtime.tools.STM32Tools.path}` in `{runtime.hardware.path}` ?

When I install the package from Arduino Manager board,  `tools/` is under `STM32/` and not `STM32/hardware/`...

I rather to ask before to fix it. Just to be sure.

fpistm, 2017-05-24:
Hi @fprwi6labs,
There is no issue.
On the repo the path is {runtime.hardware.path} in order to be able to use the Arduino_Tools repo directly.
It is change to {runtime.tools.STM32Tools.path} by the packager script in order to be inline with Arduino_Tools package.

fprwi6labs, 2017-05-24:
My bad.
I close this issue that is not one.

closed 2017-05-24 by fprwi6labs