stm32duino/Arduino_Core_STM32 #4: micros() is going backwards

danieleff, 2017-05-30:
version 2017.5.12

```
Serial.println(micros());
Serial.println(micros());
Serial.println(micros());
```
result:
```
2001996
2001965
2001939
```
expected: numbers going up

fpistm, 2017-05-30:
correction tested with:
```
void setup() {
  Serial.begin(9600);

  while (!Serial);
  Serial.println(micros());
  Serial.println(micros());
  Serial.println(micros());
  Serial.println(micros());
  Serial.println(micros());
  delay(1);
  Serial.println(micros()); 
  delay(10);
  Serial.println(micros());
  delay(100);
  Serial.println(micros());
  delay(1000);
  Serial.println(micros());
  delay(1);
  Serial.println(micros());
  delay(10);
  Serial.println(micros());  
  delay(100);
  Serial.println(micros());
  delay(1000);
  Serial.println(micros());
  delay(10000);
  Serial.println(micros());
  delay(1);
  Serial.println(micros());
  delay(10);
  Serial.println(micros());  
  delay(100);
  Serial.println(micros());
  delay(1000);
  Serial.println(micros());
  delay(1);
  Serial.println(micros());  
}
```
result:

```
micros          delta (µs)     delta (ms)
1028
1051            23	
1070            19	
1090            20	
1109            19	
2001            892           0,892
12001           10000         10
112001          100000        100
1112001         1000000       1000
1113001         1000          1
1123001         10000         10
1223001         100000        100
2223001         1000000       1000
12223001        10000000      10000
12224001        1000          1
12234001        10000         10
12334001        100000        100
13334001        1000000       1000
13335001        1000          1
```

fpistm, 2017-06-04:
Issue not fully fixed:
http://www.stm32duino.com/viewtopic.php?f=51&t=2020&start=10#p27101
Thanks Pito

commit 61924cc0 (2017-05-30): Fix #4 micros() is going backwards
commit 2fa0645d (2017-06-06): Fix #4 micros() is going backwards

Previous fix was not complete as it do not manage millisecond change.
Thanks Pito:
http://www.stm32duino.com/viewtopic.php?f=51&t=2020&start=10#p27101
closed 2017-06-06 by fpistm