stm32duino/Arduino_Core_STM32 #7: HAL callback functions in the core should should be declared as weak so they can be overridden in the sketch

fpistm, 2017-06-02:
_From @rogerclarkmelbourne on November 23, 2016 21:2_

HAL callback functions in the core should should be declared as weak so they can be overridden in the sketch

_Copied from original issue: stm32duino/Arduino_Core_STM32L4#8_

fpistm, 2017-06-02:
_From @fprwi6labs on January 10, 2017 12:22_

Can you declare the same function weak more than one time? 
In the doubt I tried and I can say that it is not possible because gcc returns an error (multiple definition).

fpistm, 2017-06-02:
_From @rogerclarkmelbourne on January 10, 2017 21:39_

OK.
Perhaps there need to be some code changes so they are only declared once as weak.

I'm sure someone else posted about this to the forum, I'll need to find the post and link it here.

fpistm, 2017-06-02:
_From @fprwi6labs on January 16, 2017 16:15_

I'm really curious to know how the linker manages the case where the same function is declared twice as weak. Maybe there is a way but I don't know it...

Adminius, 2017-06-30:
Does it mean, that i can implement interrupt driven uart receive function?

fpistm, 2017-10-20:
@Adminius you could have a look at the [SerialEvent ](https://www.arduino.cc/en/Reference/SerialEvent) function which is weak.
It is not an interrupt CB but it allows to handle received data.

Note: this will be functional in the next core release.

fpistm, 2017-10-20:
Close this issue as no update and already several weak function.
New issue could be opened if specific one are requested.

closed 2017-10-20 by fpistm