stm32duino/Arduino_Core_STM32 #9: Firmata sketch compilation failed for all boards

jmchiappa, 2017-06-04:
Hi Frédéric,

I'm using intensively firmata sketch. I've upgraded with the latest package my arduino IDE.
But now, the compilation fails for all MCU/Nucleo boards.
Here is the log extracted from arduino's IDE (in french, sorry) targetting the STM32L053R8:
```
/home/vs1/Applications/arduino-1.8.2/arduino-builder -dump-prefs -logger=machine -hardware /home/vs1/Applications/arduino-1.8.2/hardware -hardware /home/vs1/.arduino15/packages -tools /home/vs1/Applications/arduino-1.8.2/tools-builder -tools /home/vs1/Applications/arduino-1.8.2/hardware/tools/avr -tools /home/vs1/.arduino15/packages -built-in-libraries /home/vs1/Applications/arduino-1.8.2/libraries -libraries /home/vs1/Arduino/libraries -fqbn=STM32:stm32:Nucleo_64:Nucleo_64_board=NUCLEO_L476RG,upload_method=MassStorageMethod -ide-version=10802 -build-path /tmp/arduino_build_25221 -warnings=none -build-cache /tmp/arduino_cache_431276 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/vs1/.arduino15/packages/STM32/tools/STM32Tools/2017.5.12 -prefs=runtime.tools.CMSIS.path=/home/vs1/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/vs1/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update -verbose /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Firmata/examples/StandardFirmata/StandardFirmata.ino
/home/vs1/Applications/arduino-1.8.2/arduino-builder -compile -logger=machine -hardware /home/vs1/Applications/arduino-1.8.2/hardware -hardware /home/vs1/.arduino15/packages -tools /home/vs1/Applications/arduino-1.8.2/tools-builder -tools /home/vs1/Applications/arduino-1.8.2/hardware/tools/avr -tools /home/vs1/.arduino15/packages -built-in-libraries /home/vs1/Applications/arduino-1.8.2/libraries -libraries /home/vs1/Arduino/libraries -fqbn=STM32:stm32:Nucleo_64:Nucleo_64_board=NUCLEO_L476RG,upload_method=MassStorageMethod -ide-version=10802 -build-path /tmp/arduino_build_25221 -warnings=none -build-cache /tmp/arduino_cache_431276 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/vs1/.arduino15/packages/STM32/tools/STM32Tools/2017.5.12 -prefs=runtime.tools.CMSIS.path=/home/vs1/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/vs1/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update -verbose /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Firmata/examples/StandardFirmata/StandardFirmata.ino
Using board 'Nucleo_64' from platform in folder: /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2
Using core 'arduino' from platform in folder: /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2
Detecting libraries used...
"/home/vs1/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update/bin/arm-none-eabi-g++" -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/avr" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/stm32" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Inc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Src/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/STM32L4xx/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG/usb" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=4000000L -DSTM32L4xx -DARDUINO=10802 -DARDUINO_NUCLEO_L476RG -DARDUINO_ARCH_STM32  -DSTM32L476xx -DUSBD_VID=0x0483 -DUSBD_PID=0x5711 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="NUCLEO-L476RG"' "-I/home/vs1/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG" "/tmp/arduino_build_25221/sketch/StandardFirmata.ino.cpp" -o "/dev/null"
"/home/vs1/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update/bin/arm-none-eabi-g++" -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/avr" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/stm32" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Inc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Src/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/STM32L4xx/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG/usb" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=4000000L -DSTM32L4xx -DARDUINO=10802 -DARDUINO_NUCLEO_L476RG -DARDUINO_ARCH_STM32  -DSTM32L476xx -DUSBD_VID=0x0483 -DUSBD_PID=0x5711 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="NUCLEO-L476RG"' "-I/home/vs1/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Servo" "/tmp/arduino_build_25221/sketch/StandardFirmata.ino.cpp" -o "/dev/null"
"/home/vs1/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update/bin/arm-none-eabi-g++" -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/avr" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/stm32" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Inc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32L4xx_HAL_Driver/Src/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/STM32L4xx/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG/usb" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -mcpu=cortex-m4 -DF_CPU=4000000L -DSTM32L4xx -DARDUINO=10802 -DARDUINO_NUCLEO_L476RG -DARDUINO_ARCH_STM32  -DSTM32L476xx -DUSBD_VID=0x0483 -DUSBD_PID=0x5711 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="NUCLEO-L476RG"' "-I/home/vs1/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_L476RG" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Servo" "-I/home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Wire" "/tmp/ardu [... cut ...]
Utilisation de la bibliothèque Wire version 1.0 dans le dossier: /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Wire 
Utilisation de la bibliothèque Firmata version 2.5.2 dans le dossier: /home/vs1/.arduino15/packages/STM32/hardware/stm32/2017.6.2/libraries/Firmata 
exit status 1
Erreur de compilation pour la carte Nucleo-64
```
It seems that PeripheralPins.c is not evaluated before failing, even if you've declared 
`extern const PinMap PinMap_ADC[];
`
in PeripheralPins.h

Could you kindly correct and release a patch quickly ?
Thank you,
Jean-Marc

fpistm, 2017-06-04:
Hi, you're right. I have to update the lib. It is in my task list.

jmchiappa, 2017-06-05:
Hi Frédéric,
Thank you for the fix. it compiles and links now.
But it seems that something is broken in Uart Rx callback or IRQn, because my firmata client (PC side) application got the first character (Reset frame from firmware) but that's all. No other information is retrieved from my client.

you can test with this firmata client (no installation required) : 
[http://www.firmata.org/wiki/Main_Page](url)

Thank you again

Jean-Marc

PS : I'm trying to create an eclipse project from your package so that it would be easier to debug :-)

fpistm, 2017-06-05:
Thanks for the test. I will debug it tomorrow. I'm OOO today. To debug I'm using gdb through ST Link.

fpistm, 2017-06-05:
After checking the code, I think I have found the issue.
In Boards.h:
replace:
#define IS_PIN_DIGITAL(p)       (p < TOTAL_PINS)
by
#define IS_PIN_DIGITAL(p)       ((p != PIN_SERIAL_RX) && (p != PIN_SERIAL_TX) &&(p < TOTAL_PINS)

else the systemResetCallback() will reset pin as Digital IO

fprwi6labs, 2017-06-05:
@fpistm 
I confirm your modification fixes the problem of communication. 
It is OK with F4 but not with the others (L0, L4...). I'm on it.

jmchiappa, 2017-06-05:
I confirm that it doesn't work for L476RG.
It's strange; some data is exchanged (200 bytes) then.. the communication failed for some mysterious reasons... (sorry for the late reply, I'm OoO too :-) ).

fpistm, 2017-06-09:
Some update,
I've tried the updated version of the Firmata Arduino library.
Same behavior except I saw the full list of pins for F429 with the firmata client you provide.
I assume the numbers of pins could bring some issues.
More over it seems there is an issue around the ADC management, probably link to the delta btw number of ADC pin available in the PinMap_ADC (20declared)  vs the NUM_ANALOG_INPUTS (6) which represent the A0-A5.

fpistm, 2017-06-24:
Hi @jmchiappa,
I'm currently working on this issue, that's why i've opened this issue https://github.com/stm32duino/Arduino_Core_STM32/issues/37
I still have some bug with ADC but it's work better, I have some analog values now.
When all will be corrected I will update the pull and probably make a pull to the firmata/arduino github.

fpistm, 2017-06-27:
@jmchiappa ,

Firmata update done and tested on all variants.

closed 2017-06-28 by fpistm

diond31, 2017-08-03:
@jmchiappa I would like to use Eclipse + GCC + STM32duino for development.  Were you able to create an Eclipse project (mentioned above) that uses STM32duino with Eclipse + GCC?  Can you provide instructions and/or pointers on getting this working?

Also you mentioned above you were using Eclipse for debugging your STM32duino based project with the Firmata library.  What did you use for debugging (e.g. openOCD + GDB)?

I think your feedback and experience would be great for the STM32duino community knowledge base, as Eclipse is a much better tool for coding and debugging ARM projects.

Thanks you,
Dion

fpistm, 2017-08-03:
Hi @diond31, have you take a look on the [stm32duino](http://stm32duino.com/viewforum.php?f=41) forum?
Theres a lot of discussion about eclipse and plugins:
[Debugging Arduino sketch with STM32 Nucleo in Eclipse .. works !](http://stm32duino.com/viewtopic.php?f=41&t=522)
[Getting Eclipse to work? A proper guide?](http://stm32duino.com/viewtopic.php?f=41&t=2384)
And so on the ide section

commit 8d0c06de (2017-06-28): Fix Firmata library for STM32

Fix #9
All variants tested thanks:
http://www.firmata.org/wiki/Main_Page#Firmata_Test_Program