stm32duino/Arduino_Core_STM32 #12: 2017.6.2 SD libraries compilation issue

rreicher, 2017-06-06:
In any exemple which use SD (SD or TFT libraries) cannot find sd.conf.h.

C:\Users\rom\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2017.6.2\libraries\SD\src/bsp_sd.h:47:21: fatal error: sd_conf.h: No such file or directory_

Thanks.

fpistm, 2017-06-06:
Hi @rreicher,
on which target you used the SD example ?
The modified SD library is usable for STM32F7 disco which have an SD card slot.
As mentioned here https://github.com/stm32duino/Arduino_Core_STM32/issues/2

> 1- Rename the SD.h from SD library provided in this repo. By this way, including SD.h will use the standard Arduino library. Used for Arduino shield including SD card slot by through SPI.

So, if you want use the SD for Arduino shield (through SPI), you should rename the SD.h file from the modified SD library.
It will be corrected soon.

rreicher, 2017-06-07:
Hi fpistm.

I use target NucleoL476 and ST7735 1.8inch tft lcd which include a SD socket (through SPI)

From provided library the SD.h file should renamed to what ?

Thanks and best regards.

fpistm, 2017-06-07:
Ok.
So rename SD.h as you want. This will unblock your issue and use the Standard library through SPI for your shield.

rreicher, 2017-06-07:
Ok, I wil try it.
Thanks

fpistm, 2017-06-08:
This will be corrected with: https://github.com/stm32duino/Arduino_Core_STM32/issues/2

closed 2017-06-08 by fpistm