stm32duino/Arduino_Core_STM32 #13: UART_OPTIMIZED flag seems to be not correctly used

jmchiappa, 2017-06-06:
Hi all,
in uart.c file, UART_OPTIMIZED is defined and set to 0 in order to not use the optimization.
But all tests are based on #ifdef UART_OPTIMIZED, and of course, the evaluation of the expression is true (even if UART_OPTIMIZED is set to 0, the UART_OPTIMIZED flag is defined).

When I replace
`#define UART_OTPIMIZED 0`
by
`#undef UART_OPTIMIZED`
the behavior is really strange. UART Rx callback fall in error 0x08 or 0x05 sometimes.
`
void HAL_UART_ErrorCallback(UART_HandleTypeDef *huart)
{
  uart_id_e uart_id = get_uart_id_from_handle(huart);
  if(NB_UART_MANAGED == uart_id) {
    return;
  }
  printf(">>UART Error callback uart id %i : ErrorCode=0x%x\n\r",uart_id,huart->ErrorCode);
  while(1);
  uart_deinit(uart_id);
  uart_init(uart_id, g_UartHandle[uart_id].Init.BaudRate);
}
`
Do you have any idea why the code falls in error ?

Thank you,
Jean-Marc

fpistm, 2017-06-07:
Hi @jmchiappa ,
Since this commit:
[](https://github.com/stm32duino/Arduino_Core_STM32/commit/61a2fd856c14fa701fc4a17944ac6dd1eec67cee#diff-825f8b96598f3542ec96a1ba1124892fL77)
UART_OTPIMIZED is no more use at least for this core.
On which repo you work?
Thanks in advance

jmchiappa, 2017-06-07:
I'm still working on 2017.1.19... Okay,  I note that this one is deprecated.
But until the last one 2017.6.2 will be fixed for firmata, I'm using the
first one.
Anyway, I will have a look at the last one... :-)

Thank you.
Jean-Marc

closed 2017-06-07 by jmchiappa