stm32duino/Arduino_Core_STM32 #20: Merge Arduino_Core_STM32F1

fpistm, 2017-06-08:
https://github.com/stm32duino/Arduino_Core_STM32F1 have to be merged in this repo.

CMSIS ST and HALfrom STM32Cube_FW_F1 already included in this repo.

RickKimball, 2017-06-22:
I noticed you had a schedule for one of the other Issues * the low power one * and you mention a plan for week 43 on that one.  Is the F1 Core merge on the schedule with an ETA?

fprwi6labs, 2017-06-23:
@RickKimball 

The F1 core is planed for the next week. 
Be ready for a big test session. We need tester like you ;)

RickKimball, 2017-06-23:
Which boards? NUCLEO-F103RB + Bluepill? * I don't have a NUCLEO

fprwi6labs, 2017-06-23:
NUCLEO-F103RB + BluePill + MapleMini + STM32vldiscovery. All variant currently available.
I perform tests on nucleo. I have a Bluepill too but if you can do some on maple and discovery it will be perfect.

rogerclarkmelbourne, 2017-06-23:
I have the Maple Mini, so I will test during the weekend

fpistm, 2017-06-29:
Thanks @fprwi6labs

Adminius, 2017-06-29:
do you have special sketches/hw=extension board for testing?

fprwi6labs, 2017-06-30:
No, I haven't. I only use the example sketches.

fpistm, 2017-08-02:
List of boards to merge:
- [ ]     NUCLEO-F103RB
- [ ]     DISCO-F100RB (STM32VLDISCOVERY)
- [ ]     BLUEPILL-F103C8
- [ ]     MAPLEMINI-F103CB