stm32duino/Arduino_Core_STM32 #23: Unable to upload using massStorageCopy

RickKimball, 2017-06-10:
I'm creating a new variant for the NUCLEO-F030R8. When using the massStoraryCopy build, platform.txt uses the {node} parameter of the board to decide where to copy the bin file at upload time. This fails for me when I follow the scheme used by all the other board.txt entries.

On my 64 bit linux 16.04, my NUCLEO-F030R8 board shows up with the name mass storage name of "NUCLEO" instead of "NODE_F030R8".

I tried running Win7 in a VM and it also shows up as "NUCLEO" there.  

Is there an expected driver or udev entry I'm supposed to be using? Or should the my board entry look like this:
```
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node=NUCLEO
```
BTW: that works properly

fpistm, 2017-06-11:
Yes you have to set it with NUCLEO as displayed by the host os.

fpistm, 2017-06-13:
As I have mentioned in other thread.
The mount point of my F030R8 is:
`/dev/sdc on /media/<login>/NODE_F030R8 type vfat (rw,nosuid,nodev,uid=175250,gid=175250,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)`

fpistm, 2017-06-14:
This issue was moved to stm32duino/Arduino_Tools#6

closed 2017-06-14 by fpistm