stm32duino/Arduino_Core_STM32 #24: SPI clock acting weird

danieleff, 2017-06-11:
I was trying Adafruit ILI9341, but was not working.
I think the following code is a simplified version of the problem:
```
void setup() {
  int cs = 2;
  SPI.begin();
  pinMode(cs, OUTPUT);
  digitalWrite(cs, HIGH);

  while(1) {
      SPI.beginTransaction(SPISettings(1000000, MSBFIRST, SPI_MODE0));
      delay(100);
      digitalWrite(cs, LOW);
      SPI.transfer(0x01);
      digitalWrite(cs, HIGH);
      SPI.endTransaction();
      delay(100);
  }
}
````
With arduino UNO, I get the following:
[image]
With F746 discovery:
[image]
or ( with F407 always)
[image]
Logic analizer settings are the same.
Can you check if you get the same results?

fpistm, 2017-06-11:
Hi @danieleff ,

have you made those test with or without this fix?
https://github.com/stm32duino/Arduino_Core_STM32/pull/19

danieleff, 2017-06-11:
With that fix:
```
> git show
commit 055864fb16457cd45b638f48588b85477f0eb226
Merge: dadfba3 052443e
Author: Frederic Pillon <frederic.pillon@st.com>
Date:   Fri Jun 9 17:36:17 2017 +0200
```
(Actually does not compile on 407 disco, `timer.c:466:22: error: 'RCC_PeriphCLKInitTypeDef {aka struct <anonymous>}' has no member named 'TIMPresSelection'`, but its another issue, I commented out)

fpistm, 2017-06-12:
I do not check yet, but seems as GPIO pull config issue.

fpistm, 2017-06-15:
Thanks @cparata 
Pull setting on the SCK pin should be in PULLDOWN if the SPI polarity is LOW and in PULLUP if the SPI polarity is HIGH.

commit 5f4cbddd (2017-06-15): Fix SPI clock polarity issue

PULLDOWN or PULLUP the SCK pin according the polarity used.

Fix #24
closed 2017-06-15 by fpistm