stm32duino/Arduino_Core_STM32 #30: AnalogInOutSerial max reading seems low

RickKimball, 2017-06-13:
I'm using the Nucleo_F030R8 branch to test out the analog features. I used a unmodified version of the sketch. I connected pin A0 to the 3.3V pin using a 4.9k resistor. I would expect the sensor reading to be 1023. However I'm seeing 1005.
```
sensor = 1005	 output = 250
sensor = 1005	 output = 250
sensor = 1006	 output = 250
sensor = 1005	 output = 250
```
The voltage between GND and the 3.3v pins is 3.3158v

Is there a reason this isn't reading 1023?

http://tinyurl.com/ydzd2xhx

RickKimball, 2017-06-13:
FWIW: I compiled the ADC demo from the STM32Cube/Repository

~/STM32Cube/Repository/STM32Cube_FW_F0_V1.8.0/Projects/STM32F030R8-Nucleo/Examples/ADC/ADC_DMA_Transfer

It returns 4095 ( it is a 12 bit resolution example ) and this is what I would expect. So it seems the hardware is capable.

RickKimball, 2017-06-13:
If I add in a calibration routine it now produces the full range:
```
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
sensor = 1023	 output = 255
```
Here is the change I made
```
diff --git a/cores/arduino/stm32/analog.c b/cores/arduino/stm32/analog.c
index d4c66e5..f2735cd 100644
--- a/cores/arduino/stm32/analog.c
+++ b/cores/arduino/stm32/analog.c
@@ -611,6 +611,12 @@ uint16_t adc_read_value(PinName pin)
     /* ADC Calibration Error */
     return 0;
   }
+#elif defined (STM32F0xx)
+  if (HAL_ADCEx_Calibration_Start(&AdcHandle) !=  HAL_OK)
+  {
+    /* ADC Calibration Error */
+    return 0;
+  }
 #endif
 
   /*##-3- Start the conversion process ####################*/
```

fpistm, 2017-06-14:
Nice shot.
I've missed this when I merged.
https://github.com/stm32duino/Arduino_Core_STM32F0/blob/master/cores/arduino/libstm32f0/analog.c#L578

commit 3299a739 (2017-06-14): Missing ADC calibration for STM32F0

Fix #30
closed 2017-06-14 by fpistm