stm32duino/Arduino_Core_STM32 #31: unknown chip id! 0x10076415

fpistm, 2017-06-14:
_From @kuobakuoba on June 14, 2017 6:50_

I connected my STM32L476RG to the arduinoIDE with according settings.
[image]
And upload a Blink example from the IDE.

But I get these errors.
[image]
[image]

I am running mac with sierra.
[image]
I have eclipse with Ac6 for STM32 and it works with no problem.
How can I get STM32duino to work?
Thanks.

_Copied from original issue: stm32duino/wiki#1_

fpistm, 2017-06-14:
Hi @kuobakuoba,
Please, Could you try to update the STLink firmware.

[ST-LINK, ST-LINK/V2, ST-LINK/V2-1 firmware upgrade ](https://my.st.com/content/my_st_com/en/products/embedded-software/development-tool-software/stsw-link007.license%3d1497338034863.html)

kuobakuoba, 2017-06-14:
Hi!! @fpistm 
I've tried but with no luck.
[image]
It was exactly the same error code as above.
Thanks.

fpistm, 2017-06-14:
Ok thanks for the test.
Probably that the stlink binary released do not support this boards or have an issue.
Could you test with the latest version 1.3.0:
https://github.com/texane/stlink/releases/tag/1.3.0

To test, use this command line, where binary file is the path to your bin built with arduino (See the build log to find the fullpath):
stlink/st-flash write <binary path> 0x8000000

If it's work you could replace the binary installed by STM32Tools.
You also have the path in the build log:
/User/...../packages/STM32/tools/STM32Tools/2017.5.12/tools/macosx/stlink/

Unfortunately I do not have Mac to perform the test.
Maybe @rogerclarkmelbourne  has some clues about that?

kuobakuoba, 2017-06-14:
@fpistm Sorry for the late reply.
To be honest,I don't know how to use these binaries.
But with some lucks or something I managed to get it work with the file you gave me.
I just clicked st-flash and copy all file in the bin folder to here(/User/...../packages/STM32/tools/STM32Tools/2017.5.12/tools/macosx/stlink/) and for some reason it just worked.
Thank you so much for your help.

fpistm, 2017-06-14:
Ok, so if its work with the updated file, I will do an update in the [Arduino_Tools](https://github.com/stm32duino/Arduino_Tools)

fpistm, 2017-06-14:
Moved to https://github.com/stm32duino/Arduino_Tools/issues/7

closed 2017-06-14 by fpistm