stm32duino/Arduino_Core_STM32 #33: analogWrite() doesn't use the full range

RickKimball, 2017-06-16:
On a NUCLEO-F030R8, if I call analogWrite(9,255), I'm seeing a 3.9us low pulse each period. I would have expected that pin 9 should just say HIGH using a value of 255 as that is the max value for the range (0-255).  Shouldn't it do a digitalWrite(9,HIGH) if the value provided is the max of the analogResolution?

RickKimball, 2017-06-17:
It appears like the period might be off
```
diff --git a/cores/arduino/stm32/analog.c b/cores/arduino/stm32/analog.c
index 25b74b9..e96e639 100644
--- a/cores/arduino/stm32/analog.c
+++ b/cores/arduino/stm32/analog.c
@@ -719,7 +719,7 @@ void pwm_start(PinName pin, uint32_t clock_freq,
   timHandle.Instance               = pinmap_peripheral(pin, PinMap_PWM);
   if (timHandle.Instance == NC) return 0;
   timHandle.Init.Prescaler         = (uint32_t)(getTimerClkFreq(timHandle.Instance) / clock_freq) - 1;
-  timHandle.Init.Period            = period;
+  timHandle.Init.Period            = period-1;
   timHandle.Init.ClockDivision     = TIM_CLOCKDIVISION_DIV1;
   timHandle.Init.CounterMode       = TIM_COUNTERMODE_UP;
 #ifndef STM32L0xx
```
That seems to produce better results

fpistm, 2017-06-18:
Hi @RickKimball , i will check this tuesday. Thanks for the feedback

fpistm, 2017-06-20:
You are right, Period need to be decrease by 1 ( = Auto-reload value : TIMx->ARR)

commit 2d65501a (2017-06-20): ARR register have to be set at Period-1.

Fix #33
closed 2017-06-20 by fpistm