stm32duino/Arduino_Core_STM32 #34: support nucleo-l432kc (STM32L432)

Jorgen-VikingGod, 2017-06-21:
Please add support for STM32L432 like the [Nucleo-L432KC](http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-l432kc.html)

Here is another arduino port of STM32L4xx core:
https://github.com/GrumpyOldPizza/arduino-STM32L4/tree/master/variants/STM32L432KC-NUCLEO

RickKimball, 2017-06-26:
Why don't you just use that core @Jorgen-VikingGod ?

Jorgen-VikingGod, 2017-06-26:
because most of the Arduino libs are not ported.

fpistm, 2017-06-26:
Hi @Jorgen-VikingGod,
I will add it soon and take it probably  as an example for the wiki page for adding a variant.

Jorgen-VikingGod, 2017-06-27:
thx @fpistm this would be awesome! Makes it a little easier for other to add new variants in future.

Adminius, 2017-07-02:
wiki is a nice idea. 
I need support for [NUCLEO F401RE](http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f401re.html)... i would try to add it :)

Adminius, 2017-07-03:
Assumption: 
to add new board/MCU support, this files
`system\Drivers\STM32F3xx_HAL_Driver\Src\`
need to be checked, if those variant is available.
My understanding is: it should not be a whole name, e.g. STM32F401RE, because 
F4 is family = main functions
F4xx is "specials functions" ? (like CAN?)
R is 64pins
E is ROM

Make it sense to made 
`"#define ROM_E" or "STM32xxxxxE"` instead of `STM32F303xE, STM32F302xE... `
`"#define ROM_8"` instead of 
```
"#if defined(STM32F303x8) || defined(STM32F334x8) || defined(STM32F328xx) || \
    defined(STM32F301x8) || defined(STM32F302x8) || defined(STM32F318xx)"
```
and so on for all digits?
so we don't need to edit all of driver files and "ROM_8" is defined for all families.

fpistm, 2017-07-04:
@Adminius,
It's look correct
I think you should post your request to  [community.st.com](https://community.st.com/community/stm32-community)
This repo not aims to modify HAL only use them. I update HAL when new CubeFw is released.

closed 2017-07-05 by fpistm

fpistm, 2017-07-12:
Wiki created:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

PR #53 (2017-07-05): Nucleo l432kc
Variant for Nucleo L432KC
Fix #34 
To flash with STLink under Windows , ST-Link CLI v 3.0.0.0 is required:
https://github.com/stm32duino/Arduino_Tools/issues/12