stm32duino/Arduino_Core_STM32 #36: Missing node value in boards.txt for early NUCLEO-F030R8 boards

RickKimball, 2017-06-22:
To support the older F030R8 boards, the "node" entry in boards.txt needs to look like this:

boards.txt:
```
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.node=NODE_F030R8,NUCLEO
```

fpistm, 2017-06-22:
> This is probably true for all boards?
I need to check.

closed 2017-06-22 by fpistm

RickKimball, 2017-06-24:
On linux with the quotes, I get this message:
```
/home/kimballr/.arduino15/packages/STM32/tools/STM32Tools/2017.5.12/tools/linux/massStorageCopy -I /tmp/arduino_build_547147/AnalogInOutSerial.ino.bin -O "NODE_F030R8,NUCLEO" 
"NODE_F030R8 not found. please ensure the device is correctly connected
An error occurred while uploading the sketch
```
This needs to be changed as per my comment or a bug needs to be created for the massStorageCopy code so it can handle the " character

commit c99a5833 (2017-06-22): Update node entry for older F030R8 boards

Older F030R8 boards have "NUCLEO" as label and not NODE_F030R8.
MassStorageCopy tools has been updated to manage several node name.
Node names list has to be separated by ',' and double quoted (for Windows).
ex:
"Node1,Node2,Node3"

Fix #36