stm32duino/Arduino_Core_STM32 #41: Use recipe.hooks feature of the platform.txt file

fprwi6labs, 2017-06-27:
Orignally posted in #38 by @RickKimball

I think you could take advantage of the recipe.hooks feature of the platform.txt file to conditionally override defines and gcc command line arguments on a per sketch basis.  This seems to be a fairly new feature (since Arduino 1.6.something) something they added since I last looked to accomplish something like this.

You might approach it like this:

In the stm32 tools directory create a new tool called 'cpif' which copies a file called gcc_options.txt into the build directory if the file exists in the sketch directory or copies /dev/null if it doesn't
```
$ cat cpif 
#!/bin/bash
set -x
echo $0 $*

sketchdir=${1%.*}
dest=${2}/gcc_options.txt

options="$HOME/Arduino/$sketchdir/gcc_options.txt"

if [ -f $options ]
then
   cp $options $dest
else
   cp /dev/null $dest
fi
```
In platform.txt you add a recipe hook that is called before starting the build process.
```
 ## Compile c files
+recipe.hooks.prebuild.1.pattern=cpif {build.project_name} {build.path} b4all
 recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mcpu={build.mcu} -DF_CPU={build ...
```
modify the board.txt to use the @file feature of gcc or just add it to the platform.txt for all boards
```
Nucleo_64.menu.Nucleo_64_board.NUCLEO_F030R8.build.extra_flags=-DSTM32F030x8 @{build.path}/gcc_options.txt
```
When the user wants to override defines and gcc command line arguments they create a file called gcc_options.txt in the sketch directory. Here I imagine I want to override the F_CPU value
```
-U F_CPU
-D F_CPU=24000000UL
```
When I compile it uses the sketch specific defines and overrides.
```
/home/kimballr/github/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware /home/kimballr/github/Arduino/build/linux/work/hardware -hardware /home/kimballr/.arduino15/packages -hardware /home/kimballr/Arduino/hardware -tools /home/kimballr/github/Arduino/build/linux/work/tools-builder -tools /home/kimballr/github/Arduino/build/linux/work/hardware/tools/avr -tools /home/kimballr/.arduino15/packages -built-in-libraries /home/kimballr/github/Arduino/build/linux/work/libraries -libraries /home/kimballr/Arduino/libraries -fqbn=STM32:stm32:Nucleo_64:Nucleo_64_board=NUCLEO_F030R8,upload_method=MassStorageMethod -ide-version=10803 -build-path /tmp/arduino_build_433355 -warnings=more -build-cache /tmp/arduino_cache_805852 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/kimballr/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/kimballr/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update -prefs=runtime.tools.STM32Tools.path=/home/kimballr/.arduino15/packages/STM32/tools/STM32Tools/2017.5.12 -verbose /home/kimballr/Arduino/BlinkWithoutDelay_HSE_Bypassx/BlinkWithoutDelay_HSE_Bypassx.ino
/home/kimballr/github/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware /home/kimballr/github/Arduino/build/linux/work/hardware -hardware /home/kimballr/.arduino15/packages -hardware /home/kimballr/Arduino/hardware -tools /home/kimballr/github/Arduino/build/linux/work/tools-builder -tools /home/kimballr/github/Arduino/build/linux/work/hardware/tools/avr -tools /home/kimballr/.arduino15/packages -built-in-libraries /home/kimballr/github/Arduino/build/linux/work/libraries -libraries /home/kimballr/Arduino/libraries -fqbn=STM32:stm32:Nucleo_64:Nucleo_64_board=NUCLEO_F030R8,upload_method=MassStorageMethod -ide-version=10803 -build-path /tmp/arduino_build_433355 -warnings=more -build-cache /tmp/arduino_cache_805852 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=/home/kimballr/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/kimballr/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update -prefs=runtime.tools.STM32Tools.path=/home/kimballr/.arduino15/packages/STM32/tools/STM32Tools/2017.5.12 -verbose /home/kimballr/Arduino/BlinkWithoutDelay_HSE_Bypassx/BlinkWithoutDelay_HSE_Bypassx.ino
Using board 'Nucleo_64' from platform in folder: /home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2
Using core 'arduino' from platform in folder: /home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2
cpif BlinkWithoutDelay_HSE_Bypassx.ino /tmp/arduino_build_433355 b4all+ echo /home/kimballr/opt64/bin/cpif BlinkWithoutDelay_HSE_Bypassx.ino /tmp/arduino_build_433355 b4all

+ sketchdir=BlinkWithoutDelay_HSE_Bypassx
/home/kimballr/opt64/bin/cpif BlinkWithoutDelay_HSE_Bypassx.ino /tmp/arduino_build_433355 b4all
+ dest=/tmp/arduino_build_433355/gcc_options.txt
+ options=/home/kimballr/Arduino/BlinkWithoutDelay_HSE_Bypassx/gcc_options.txtDetecting libraries used...

+ '[' -f /home/kimballr/Arduino/BlinkWithoutDelay_HSE_Bypassx/gcc_options.txt ']'
+ cp /home/kimballr/Arduino/BlinkWithoutDelay_HSE_Bypassx/gcc_options.txt /tmp/arduino_build_433355/gcc_options.txt
"/home/kimballr/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update/bin/arm-none-eabi-g++" -mthumb -c -O2 -g -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/avr" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/stm32" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32F0xx_HAL_Driver/Inc/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32F0xx_HAL_Driver/Src/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/STM32F0xx/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_F030R8/usb" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -mcpu=cortex-m0 -DF_CPU=48000000L -DSTM32F0xx -DARDUINO=10803 -DARDUINO_NUCLEO_F030R8 -DARDUINO_ARCH_STM32  -DSTM32F030x8 @/tmp/arduino_build_433355/gcc_options.txt "-I/home/kimballr/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/gcc/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_F030R8" "/tmp/arduino_build_433355/sketch/BlinkWithoutDelay_HSE_Bypassx.ino.cpp" -o "/dev/null"
Using cached library dependencies for file: /tmp/arduino_build_433355/sketch/customclock.c
Generating function prototypes...
"/home/kimballr/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q1-update/bin/arm-none-eabi-g++" -mthumb -c -O2 -g -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/avr" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/cores/arduino/stm32" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32F0xx_HAL_Driver/Inc/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/STM32F0xx_HAL_Driver/Src/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/STM32F0xx/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/variants/NUCLEO_F030R8/usb" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -mcpu=cortex-m0 -DF_CPU=48000000L -DSTM32F0xx -DARDUINO=10803 -DARDUINO_NUCLEO_F030R8 -DARDUINO_ARCH_STM32  -DSTM32F030x8 @/tmp/arduino_build_433355/gcc_options.txt "-I/home/kimballr/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/home/kimballr/.arduino15/packages/STM32/hardware/stm32/2017.6.2/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/" "-I/home/kimballr/.arduino15/packages/S [... cut ...]
Sketch uses 7564 bytes (11%) of program storage space. Maximum is 65536 bytes.
Global variables use 4064 bytes (49%) of dynamic memory, leaving 4128 bytes for local variables. Maximum is 8192 bytes.
```
Of course you would have to make this work with the different OS types. You would also have to deal with finding the users preference sketch home directory. I just hard coded it in the example above to use my $HOME/Arduino directory.

Seems doable (I actually have this working ) if you care to open yourself to all kinds of user problems  when they do bad things : )

fpistm, 2017-06-27:
Thanks  @fprwi6labs.
You're right it is common for all libs and core (ex: enable USB)
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#pre-and-post-build-hooks-since-ide-165
Seems to be promising as we could operate at several level (core, lib, sketch...)
Thanks @RickKimball for the tips

fpistm, 2017-12-11:
Hi @RickKimball 
I've (finally) make a generic implementation based on your request. 
Let me know your feedback.

RickKimball, 2017-12-11:
I just saw your change. Excellent thanks!  I'll give it a spin.. Your idea to use a .h file is a great idea.

PR #174 (2017-12-19): Build option
It is now possible to customize some core definitions
or gcc options thanks a file named "`build_opt.h"`
in the sketch directory. (using @file gcc option)

User can add it thanks the Arduino IDE in a tab to easily customize the core options.
That's why '.h' extension is used as IDE allows only those one: ino, h, hpp, c, cpp, pde.
Note: file is automatically copied in the `{build.path}/sketch/` directory

Example of file content to change F_CPU value:
`-UF_CPU -DF_CPU=72000000UL`

If the file does not exist an empty one is created in the `{build.path}/sketch/` directory.

Fix #41
commit 130c5ab3 (2017-12-19): Added recipe.hooks feature to override gcc options

It is now possible to customize some core definitions
or gcc options thanks a file named "build_opt.h"
in the sketch directory. (using @file gcc option)

Example of file content to change F_CPU value:
-UF_CPU -DF_CPU=72000000UL

If the file does not exist an empty one is created.

Fix #41
closed 2017-12-19 by fpistm