stm32duino/Arduino_Core_STM32 #47: Problems with EEPROM.read() on L476RG

Adminius, 2017-06-29:
i've tried to use a eeprom_clear demo sketch, but if i upload it, LED stays off.

Then i tried to read a EEPROM:
```
#include <EEPROM.h>

void setup() {
  Serial.begin(115200);
  Serial.print("EEPROM Size: ");
  Serial.println(EEPROM.length()); //it works with only with delay(10);
  delay(10); 
  Serial.print("Data: ");
  Serial.println(EEPROM.read(0));
  Serial.println("I'm here"); //i've never seen this line :/
}
void loop() {}
```
Console output with delay(10);
`EEPROM Size: 16384
Dat`
without delay:
`EEP`

i use official core release 2017.06.2
Upload via Mass Storage.

fpistm, 2017-06-30:
Tested on L476:

- [ ] eeprom_clear
- [ ] eeprom_crc
- [ ] eeprom_get
- [ ] eeprom_iteration
- [ ] eeprom_put
- [ ] eeprom_read
- [ ] eeprom_update
- [ ] eeprom_write

Adminius, 2017-06-30:
thx. no more problems 👍

commit 21985caa (2017-06-30): Merge pull request #49 from fpistm/issue47_EEPROM

Fix #47 eeprom issue
commit e49849b4 (2017-06-30): Fix #47 eeprom issue
closed 2017-06-30 by Adminius