stm32duino/Arduino_Core_STM32 #48: Move Arduino libraries dedicated for STM32 core into single repo

fpistm, 2017-06-30:
Purpose of this issue is to move Arduino libraries dedicated for STM32 core in separate repository.

I talk about libraries dedicated to hardware features: on board SD, Ethernet, LCD….
For purely software libraries, pull request should be done on dedicated github (ex: firmata,…) as far as possible else a repo could be created.

The core should only embed those for common hardware support (Wire/SPI/…).
As specified here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager
So one library == a single repo.

- [ ] SD required 2 repos: 
- FatFs
- SDIOStm32
- [ ] NativeEthernet required 2 repos:
- LwIP
- EthernetStm32
- [ ] Firmata: pull request to do on related github
- [ ] CapacitiveSensor-0.5.1 to do on related github
- [ ] EEPROM; pull request to do on related github
- [ ] LiquidCrystal; pull request to do on related github
- [ ] GSM: tbd
- [ ] Mouse & Keyboard: separated repo?
- [ ] TFT: pull request to do on related github -->if possible

ARDUINO_ARCH_STM32 must be used as differentiator.

This is a first approach any comments/suggestions are welcome. Mainly on repo name or way of working.

Testato, 2017-07-05:
in my opinion open a repo for every library it is dispersive, 
The lib can remain here, in a separate folder, the important think is that all library, in the file library.proprietry, change it's name from the arduino original name. Because the ArduinoIDE have problem on update when two lib have same name.

For example it is enough call the Wire library "Wire(STM32)"

fpistm, 2017-07-05:
Firmata need an update, I will do the pull request on the official repo to add support for the STM32 then I will delete it from the core.

Open a repo for each specific library allow to use Library manager from Arduino. Having all libraries in the core is not a good option. Many libraries are specific to a board (ex: ethernet is available only on few boards (F429zi, Disco746)) so this will increase core size while user will not use it. If user wants then he will have the possibility to install it.

Testato, 2017-07-06:
It is true that you do not have ide library update, but if the lib is in the core it will updated when update the core by board manager, may be some modification on a lib needed by a core modification, in this case you can go out of sync.

Very big core peoject, like ESP, have it's modified lib inside the core. Another useful think is that when the libs is in the core you can immediately view how many lib you have so what you can do, without search for a lib without cechk if it is compatible.

But i know that also the separated github have advantage, like tracking of issue directly to a lib repo, instead of all togheter on the core repo

fpistm, 2017-08-07:
PR for Firmata merged in official repo.
Wainting new release version before removing it from the core.

fpistm, 2017-08-20:
Release of [Firmata 2.5.7](https://github.com/firmata/arduino/releases/tag/2.5.7) available.

fpistm, 2017-08-23:
TFT: pull request done: [arduino-libraries/TFT#11](https://github.com/arduino-libraries/TFT/pull/11)
Required #90

straccio, 2017-11-30:
Another libraries could be: 
- OneWire (based on HAL) 
- FreeRTOS (that's will be awesome)
- WolfSSL (really useful for IoT projects)

fpistm, 2017-11-30:
Right @straccio,
there's many libraries which could be fine to test/port.
This issue was opened for the one which was included in the core (built-in) at its beginning not to request some new library porting ;)

straccio, 2017-11-30:
Ok, Sorry!
but FreeRTOS is inside the stm32CubeFW ^^
Anyway i'll working on these libraries in the near future!

fpistm, 2017-11-30:
No worries ;)

closed 2019-02-12 by fpistm