stm32duino/Arduino_Core_STM32 #50: SERIAL_8E1: wrong data

Adminius, 2017-06-30:
```
//#define SERIAL SerialUSB //Zero
#define SERIAL Serial //STM32

void setup() {
  SERIAL.begin(115200);
  while(!SERIAL);
  Serial1.begin(19200,SERIAL_8E1);
  SERIAL.println("Setup ready");
}

void loop() {
  SERIAL.println("Send 0xFF");
  Serial1.write(0xFF);
  int temp = 0;
  delay(50);
  while(Serial1.available()>0){
      temp = Serial1.read();
      SERIAL.print("Data: ");
      SERIAL.println(temp,HEX);
  }
  delay(10000);
}
```
i tested this code on Arduino Zero and L476 and with a loop connection (RX shorted to TX)

Zero write 0xFF and read 0xFF
L476 write 0xFF and read 0x7F

fpistm, 2017-07-01:
Right,
After checking HAL uart:
  Depending on the frame length defined by the M1 and M0 bits (7-bit, 
  8-bit or 9-bit), the possible UART formats are listed in the 
  following table.
  
  Table 1. UART frame format.
```
    +-----------------------------------------------------------------------+
    |  M1 bit |  M0 bit |  PCE bit  |             UART frame                |
    |---------|---------|-----------|---------------------------------------|
    |    0    |    0    |    0      |    | SB |    8 bit data   | STB |     |
    |---------|---------|-----------|---------------------------------------|
    |    0    |    0    |    1      |    | SB | 7 bit data | PB | STB |     |
    |---------|---------|-----------|---------------------------------------|
    |    0    |    1    |    0      |    | SB |    9 bit data   | STB |     |
    |---------|---------|-----------|---------------------------------------|
    |    0    |    1    |    1      |    | SB | 8 bit data | PB | STB |     |
    |---------|---------|-----------|---------------------------------------|
    |    1    |    0    |    0      |    | SB |    7 bit data   | STB |     |
    |---------|---------|-----------|---------------------------------------|
    |    1    |    0    |    1      |    | SB | 6 bit data | PB | STB |     |
    +-----------------------------------------------------------------------+
```

In HardwareSerial.cpp:
```
void HardwareSerial::begin(unsigned long baud, byte config)
{
  _serial.baudrate = (uint32_t)baud;

  // Could be 8 or 9 bits. Could match with Arduino small data length?
  _serial.databits = UART_WORDLENGTH_8B;

  if((config & 0x30) == 0x30) {
    _serial.parity = UART_PARITY_ODD;
  } else if((config & 0x20) == 0x20) {
    _serial.parity = UART_PARITY_EVEN;
  } else {
    _serial.parity = UART_PARITY_NONE;
  }
```
if parity is set odd or even, need to set :
`    _serial.databits = UART_WORDLENGTH_9B;`

and it is ok.
I need to check all possible value provided in HardwareSerial.h, I think HAL could not deal with all those config (5bits?...):
```
#define SERIAL_5N1 0x00
#define SERIAL_6N1 0x02
#define SERIAL_7N1 0x04
#define SERIAL_8N1 0x06
#define SERIAL_5N2 0x08
#define SERIAL_6N2 0x0A
#define SERIAL_7N2 0x0C
#define SERIAL_8N2 0x0E
#define SERIAL_5E1 0x20
#define SERIAL_6E1 0x22
#define SERIAL_7E1 0x24
#define SERIAL_8E1 0x26
#define SERIAL_5E2 0x28
#define SERIAL_6E2 0x2A
#define SERIAL_7E2 0x2C
#define SERIAL_8E2 0x2E
#define SERIAL_5O1 0x30
#define SERIAL_6O1 0x32
#define SERIAL_7O1 0x34
#define SERIAL_8O1 0x36
#define SERIAL_5O2 0x38
#define SERIAL_6O2 0x3A
#define SERIAL_7O2 0x3C
#define SERIAL_8O2 0x3E
```

Adminius, 2017-07-02:
I changed to:
```
 if((config & 0x30) == 0x30) {
    _serial.parity = UART_PARITY_ODD;
    _serial.databits = UART_WORDLENGTH_9B;
  } else if((config & 0x20) == 0x20) {
    _serial.parity = UART_PARITY_EVEN;
    _serial.databits = UART_WORDLENGTH_9B;
  } else {
    _serial.parity = UART_PARITY_NONE;
    _serial.databits = UART_WORDLENGTH_8B;
  }
```

and it works (also with my HW that requires 19200 8E1).

fpistm, 2017-07-03:
Tested with:
```
typedef struct serialTest_s serialTest;
struct serialTest_s {
  char* name;
  uint32_t config;
};

static serialTest serialConfig[] = {
#ifdef UART_WORDLENGTH_7B
{"SERIAL_7N1", SERIAL_7N1},
{"SERIAL_7N2", SERIAL_7N2},
{"SERIAL_6E1", SERIAL_6E1},
{"SERIAL_6E2", SERIAL_6E2},
{"SERIAL_6O1", SERIAL_6O1},
{"SERIAL_6O2", SERIAL_6O2},
#endif
{"SERIAL_8N1", SERIAL_8N1},
{"SERIAL_8N2", SERIAL_8N2},
{"SERIAL_7E1", SERIAL_7E1},
{"SERIAL_8E1", SERIAL_8E1},
{"SERIAL_7E2", SERIAL_7E2},
{"SERIAL_7O1", SERIAL_7O1},
{"SERIAL_8O1", SERIAL_8O1},
{"SERIAL_7O2", SERIAL_7O2},
{"SERIAL_8O2", SERIAL_8O2},
{"SERIAL_8E2", SERIAL_8E2},

};

static uint32_t configCur = 0;
static uint32_t configNb = sizeof(serialConfig)/sizeof(serialTest);
static uint32_t nbTestOK = 0;
static uint32_t nbTestKO = 0;

uint32_t dataMask(uint32_t config) {
  uint32_t databits = 0;
  switch(config & 0x07) {
    case 0x02:
      databits = 6;
      break;
    case 0x04:
      databits = 7;
      break;
    case 0x06:
      databits = 8;
      break;
    default:
      databits = 0;
      break;
  }
  return ((1 << databits) - 1);
}

void test_uart(int val)
{
  int recval = 0;
  SERIAL_PORT_MONITOR.print("Send: 0x");
  SERIAL_PORT_MONITOR.print(val,HEX);
  Serial1.write(val);
  delay(10);
  while(Serial1.available()){
    recval = Serial1.read();
    SERIAL_PORT_MONITOR.print("\tReceived: 0x");
    SERIAL_PORT_MONITOR.print(recval,HEX);
  }
  if(val == recval) {
    SERIAL_PORT_MONITOR.println(" --> OK");
    nbTestOK++;
  }
  else {
    SERIAL_PORT_MONITOR.println(" --> KO <--");
    nbTestKO++;
    delay(10000);
  }
}

void setup() {
  SERIAL_PORT_MONITOR.begin(115200);
  while(!SERIAL_PORT_MONITOR);
  SERIAL_PORT_MONITOR.println("SerialLoop test");
  SERIAL_PORT_MONITOR.print(configNb);
  SERIAL_PORT_MONITOR.println(" configs to test.");
}

void loop() {
  int val = 0xFF;
  uint32_t mask = 0;
  if (configCur == configNb) {
    SERIAL_PORT_MONITOR.println("SerialLoop test done.\nResults:");
    SERIAL_PORT_MONITOR.print("OK: ");
    SERIAL_PORT_MONITOR.println(nbTestOK);
    SERIAL_PORT_MONITOR.print("KO: ");
    SERIAL_PORT_MONITOR.println(nbTestKO);
    while(1);
  }

  SERIAL_PORT_MONITOR.print("########################\nSerial1 config: ");
  SERIAL_PORT_MONITOR.print(serialConfig[configCur].name);
  SERIAL_PORT_MONITOR.print(" (0x");
  SERIAL_PORT_MONITOR.print(serialConfig[configCur].config, HEX);
  SERIAL_PORT_MONITOR.println(")");
  Serial1.begin(19200,serialConfig[configCur].config);
  mask = dataMask(serialConfig[configCur].config);
  for (int i=0; i<=(0xFF&mask); i++)
    test_uart(i&mask);

  Serial1.end();
  SERIAL_PORT_MONITOR.println("End.");
  configCur++;
}

```

PR #51 (2017-07-05): Fix Serial config
Fix #50
Arduino defined several configs (data, parity and stop bits) values
for Serial.begin(speed, config)
See: https://www.arduino.cc/en/Serial/Begin
Below configs could not be supported by STM32 as minimum
wordlength (data + parity) is UART_WORDLENGTH_7B (7 bits).
SERIAL_5N1
SERIAL_5N2
SERIAL_5E1
SERIAL_5E2
SERIAL_5O1
SERIAL_5O2
SERIAL_6N1
SERIAL_6N2
commit aa7949da (2017-07-05): Fix Serial config

Fix #50
Arduino defined several configs (data, parity and stop bits) values
for Serial.begin(speed, config)
See: https://www.arduino.cc/en/Serial/Begin
Below configs could not be supported by STM32 as minimum
wordlength (data + parity) is UART_WORDLENGTH_7B (7 bits).
SERIAL_5N1
SERIAL_5N2
SERIAL_5E1
SERIAL_5E2
SERIAL_5O1
SERIAL_5O2
SERIAL_6N1
SERIAL_6N2
closed 2017-07-05 by fpistm