stm32duino/Arduino_Core_STM32 #54: Not all STM32 Pin names (e.g. PC6,PC8...) are working

Adminius, 2017-07-05:
i tried to test PWM on 19 pins. with this code:
`int pins[] = {PC6,PC8,PC9,PB8,PB9,PA5,PA6,PA7,PB6,PC7,PA9,PA8,PB10,PB3,PA10,PB11,PB1,PB7,PB0}; 
//int pins[] = {34, 33, 32, 15, 14, 13, 12, 11, 10, 9,  8,  7,  6,   3,  2,   39,  41, 22, 49 };

void setup() {
  
}

void loop() {
  for(int i = 0; i<19;i++){
    for (int fadeValue = 0 ; fadeValue <= 255; fadeValue += 5) {
      analogWrite(pins[i], fadeValue);
      delay(30);
    }
  }
  for(int i = 0; i<19;i++){
    for (int fadeValue = 255 ; fadeValue >= 0; fadeValue -= 5) {
      analogWrite(pins[i], fadeValue);
      delay(30);
    }
  } 
}`
if i use STM32 style names, not all LEDs are working.
With Arduino names all LEDs are working.

For NUCLEO Morpho connectors it's easier to use STM32 style.

Tested on L476 with 2017.6.2 core.

fprwi6labs, 2017-07-05:
Could you indicate us which pin (STM32 name) doesn't work when you use the Arduino name?
I think PB1 can't work because all line are commented inside PinMap_PWM except you changed it.

Adminius, 2017-07-05:
PC6,PC8,PC9 are not working... the others i have to check one more time.
But the same pins, if i use 34 instead of "PC6", 33 instead of "PC8" and 32 instead of "PC9" are working.
I thing it is name resolving issue.
With a digitalWrite(PC6,HIGH) i have the same issue. But digitalWrite(33,HIGH) is working.

fpistm, 2017-07-05:
Hi @Adminius,
we aim to use Arduino naming style for better compatibility.

This is normal, using PC6  is equivalent to call with 38 (0x26).
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/PinNames.h#L73
Arduino naming Dx is an index to match PinName in an array

You could use pinNametoDigitalPin() function to convert PYx to Dx (but it seems not well exported :'( I have to fix that)

One other solution could be to provide API with PinName as argument analogWrite2(PinName)

Adminius, 2017-07-05:
Ahh, ok...
I'm woundering that some PYx names working, eg. PA9

pinNametoDigitalPin is ok for me.

fpistm, 2017-07-05:
I have correcting the issue with the pinNametoDigitalPin export.
I will push it.

So your sketch could use it like that (Serial could be removed):
```
PinName pinsSTM[] = {PC6,PC8,PC9,PB8,PB9,PA5,PA6,PA7,PB6,PC7,PA9,PA8,PB10,PB3,PA10,PB11,PB1,PB7,PB0};
int pins[] = {34, 33, 32, 15, 14, 13, 12, 11, 10, 9, 8, 7, 6, 3, 2, 39, 41, 22, 49 };

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
}

void loop() {
  for(int i = 0; i<19;i++){
  Serial.print(" pins["); Serial.print(i); Serial.print("]: "); Serial.print(pins[i]);
  Serial.print(" pinsSTM["); Serial.print(i); Serial.print("]: "); Serial.print(pinsSTM[i]);
  Serial.print(" --> "); Serial.println(pinNametoDigitalPin(pinsSTM[i]));
    for (int fadeValue = 0 ; fadeValue <= 255; fadeValue += 5) {
      analogWrite(pinNametoDigitalPin(pinsSTM[i]), fadeValue);
      delay(30);
    }
  }
  for(int i = 0; i<19;i++){
    for (int fadeValue = 255 ; fadeValue >= 0; fadeValue -= 5) {
      analogWrite(pinNametoDigitalPin(pinsSTM[i]), fadeValue);
      delay(30);
    }
  }
}
```

PR #55 (2017-07-05): Fix pinNametoDigitalPin issue
Fix #54
commit ed795137 (2017-07-05): Fix pinNametoDigitalPin issue

Fix #54
closed 2017-07-05 by fpistm