stm32duino/Arduino_Core_STM32 #56: NUCLEO F401RE support

Adminius, 2017-07-06:
first try:
https://github.com/Adminius/Arduino_Core_STM32/commit/0646d9167f8ab9755a30b91fd9275436c3d7b296

i used CubeF4 to generate some files. but i don't kwon if it was right settings

i can compile and upload, but the board do nothing. (blink doesn't blink)
any tips?

Adminius, 2017-07-07:
http://stm32duino.com/viewtopic.php?f=48&t=2304&p=31082

Adminius, 2017-07-07:
tested with auto genereted file, same issue
https://github.com/Adminius/Arduino_Core_STM32/commit/9235195e70490512397734388ce5ea24af037c25

fpistm, 2017-07-07:
Hi @Adminius,

it seems this is your clock config which is not correct.

Try this one, with it LED is blinking ;):
```
/**
  * @brief  System Clock Configuration
  *         The system Clock is configured as follow : 
  *            System Clock source            = PLL (HSI)
  *            SYSCLK(Hz)                     = 84000000
  *            HCLK(Hz)                       = 84000000
  *            AHB Prescaler                  = 1
  *            APB1 Prescaler                 = 2
  *            APB2 Prescaler                 = 1
  *            HSI Frequency(Hz)              = 16000000
  *            PLL_M                          = 16
  *            PLL_N                          = 336
  *            PLL_P                          = 4
  *            PLL_Q                          = 7
  *            VDD(V)                         = 3.3
  *            Main regulator output voltage  = Scale2 mode
  *            Flash Latency(WS)              = 2
  * @param  None
  * @retval None
  */
WEAK void SystemClock_Config(void)
{
  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_OscInitTypeDef RCC_OscInitStruct;

  /* Enable Power Control clock */
  __HAL_RCC_PWR_CLK_ENABLE();

  /* The voltage scaling allows optimizing the power consumption when the device is
     clocked below the maximum system frequency, to update the voltage scaling value
	 regarding system frequency refer to product datasheet.  */
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);

  /* Enable HSI Oscillator and activate PLL with HSI as source */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = 0x10;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 16;
  RCC_OscInitStruct.PLL.PLLN = 336;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV4;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if(HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    /* Initialization Error */
    while(1);
  }
  /* Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2
     clocks dividers */
  RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  if(HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK)
  {
    /* Initialization Error */
    while(1);
  }
  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);

  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}

```

Adminius, 2017-07-07:
should we really disable PWM on PA2/PA3 (Serial Pins)?
PWM doesn't work D0 and D1 because of ST-Link connection.
On a custom board it can be different. I sink it make sense to allow PWM on D0/D1  

my offer:

> const PinMap PinMap_PWM[] = {
> //  {PA0,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // A0  // TIM2_CH1
>     {PA0,   TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 1, 0)},  // A0  // TIM5_CH1
> //  {PA1,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 2, 0)},  // A1  // TIM2_CH2
>     {PA1,   TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 2, 0)},  // A1  // TIM5_CH2
> //  {PA2,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 3, 0)},  // D1  // TIM2_CH3  - ST-Link
>     {PA2,   TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 3, 0)},  // D1  // TIM5_CH3  - ST-Link
> //  {PA2,   TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 1, 0)},  // D1  // TIM9_CH1  - ST-Link
> //  {PA3,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 4, 0)},  // D0  // TIM2_CH4  - ST-Link
>     {PA3,   TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 4, 0)},  // D0  // TIM5_CH4  - ST-Link
> //  {PA3,   TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 2, 0)},  // D0  // TIM9_CH2  - ST-Link
>     {PA5,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // D13 // TIM2_CH1  - SPI
>     {PA6,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},  // D12 // TIM3_CH1  - SPI
> //  {PA7,   TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 1)},  // D11 // TIM1_CH1N - SPI
>     {PA7,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)},  // D11 // TIM3_CH2
>     {PA8,   TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 0)},  // D7  // TIM1_CH1
>     {PA9,   TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 0)},  // D8  // TIM1_CH2
>     {PA10,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 3, 0)},  // D2  // TIM1_CH3
>     {PA11,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 4, 0)},  // D37 // TIM1_CH4
>     {PA15,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // D21 // TIM2_CH1
> //  {PB0,   TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // A3  // TIM1_CH2N
>     {PB0,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)},  // A3 // TIM3_CH3
> //  {PB1,   TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 3, 1)},  // D41 // TIM1_CH3N
>     {PB1,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)},  // D41 // TIM3_CH4
>     {PB3,   TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 2, 0)},  // D3  // TIM2_CH2
>     {PB4,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},  // D5  // TIM3_CH1
>     {PB5,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)},  // D4  // TIM3_CH2
>     {PB6,   TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)},  // D10 // TIM4_CH1
>     {PB7,   TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 2, 0)},  // D22 // TIM4_CH2
> //  {PB8,   TIM10,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM10, 1, 0)}, // D15 // TIM10_CH1 - SCL
>     {PB8,   TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)},  // D15 // TIM4_CH3  - SCL
> //  {PB9,   TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM11, 1, 0)}, // D14 // TIM11_CH1 - SDA
>     {PB9,   TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)},  // D14 // TIM4_CH4  - SDA
>     {PB10,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 3, 0)},  // D6  // TIM2_CH3
>     {PB13,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 1)},  // D44 // TIM1_CH1N
>     {PB14,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // D43 // TIM1_CH2N
>     {PB15,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 3, 1)},  // D42 // TIM1_CH3N
>     {PC6,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},  // D34 // TIM3_CH1
>     {PC7,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)},  // D9  // TIM3_CH2
>     {PC8,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)},  // D33 // TIM3_CH3
>     {PC9,   TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)},  // D32 // TIM3_CH4
>     {NC,    NP,    0}
> };
>

fpistm, 2017-07-07:
In fact, we can do what we want.
Up to end user to know restriction.
If D0 and D1 required for other stuff then Serial instance should be change to another Pins.
I could also update firmata to ignore PWM or ADC is Serial is used. Maybe a good idea.

Adminius, 2017-07-07:
my intention: as flexible as possible.
P.S. if we should make all stuff completely idiot proof, we should also disable the same timer/channel on different pins... but i hate suchlike restrictions :/
If it possible, pls. update firmata ;) thx!

fpistm, 2017-07-07:
We are agreed ;)
I will do the update for firmata.

Adminius, 2017-07-11:
PWM OK
Serial OK
Interrups OK
EEPROM read doesn't work :(

fpistm, 2017-07-12:
I could not check the issue until begin August.
I will be in vacation for 2weeks and before leave I'm performing a release.
Maybe @fprwi6labs  or someone else could have a look.

Adminius, 2017-07-12:
It is ok for me.
Happy holidays and good weather! :)

fprwi6labs, 2017-07-12:
@Adminius 
Could you try to replace in cores/arduino/stm32/stm32_eeprom.c

> #elif defined (STM32F4xx)
// Use the last 16 page of the second bank (sector 15)
#define FLASH_BASE_ADDRESS  ((uint32_t)(0x0810C000))
#define FLASH_DATA_SECTOR   15

with

> #elif defined (STM32F4xx)
// Use the last 16 page of the second bank (sector 15)
#define FLASH_BASE_ADDRESS  ((uint32_t)(0x0806 0000))
#define FLASH_DATA_SECTOR   7

Adminius, 2017-07-12:
thx. this works.
but i can not test other supported F4xx boards...

fpistm, 2017-07-12:
It should be fine to not have FLASH_BASE_ADDRESS  and other hard coded.
I think it should be possible to use define to compute those one like I've made for L4

```
#elif defined (STM32L4xx)
#ifndef FLASH_BANK_2
#define FLASH_BANK_NUMBER   FLASH_BANK_1
#else
#define FLASH_BANK_NUMBER   FLASH_BANK_2
#endif // FLASH_BANK_2
// Flash base address
#define FLASH_PAGE_NUMBER   ((uint32_t)((FLASH_SIZE/FLASH_PAGE_SIZE) -1))
#define FLASH_BASE_ADDRESS ((uint32_t)(0x08000000 + (FLASH_PAGE_NUMBER*FLASH_PAGE_SIZE)))
```

fprwi6labs, 2017-07-13:
I know your are in vacation soon @fpistm so I will push a PR with the modification today before the new release.

fpistm, 2017-07-31:
Thanks @Adminius  for your PR. It is merged.
I've also pushes some minor corrections.

closed 2017-07-31 by fpistm