stm32duino/Arduino_Core_STM32 #67: How add a variant  from an external project

fprwi6labs, 2017-07-20:
Question from [stm32duino](http://stm32duino.com/viewtopic.php?f=48&t=2357):

Roger Clark suggestion:

> Someone on another forum recently told me about being able to add new boards to an existing core, but I have not had time to try it.
I will take a look at your files as if we can separate out the boards into different json files it will simply the number of variants in the core repo
We could also do things like bake json files for the blue pill and black pill etc etc

fpistm, 2018-01-13:
It has been done by @tomvdb 
https://github.com/tomvdb/edubot_arduino

See board.txt
https://github.com/tomvdb/edubot_arduino/blob/f895c710718ee98e10e3661f46b58e2c058e3a1d/boards.txt#L21

`BennyBot.build.core=STM32:arduino`

fpistm, 2019-06-22:
@RickKimball 
I found this on your GitHub repositories:
https://github.com/RickKimball/vendor

Can I link to this in the Wiki to answer this issue?
Thanks in advance.

RickKimball, 2019-06-23:
@fpistm .. sure

Just be warned I haven't tried that github since you added the bluepill as a official board. However, it does give someone hints at some of the things they might need to do.  Just don't expect me to answer questions on why someone's personal board doesn't work. : )

fpistm, 2019-06-23:
Sure. This is mainly for the hints and examples.

fpistm, 2019-06-28:
I've added a new Wiki page to answer this issue:
https://github.com/stm32duino/wiki/wiki/Custom-board-based-on-a-core
Thanks @RickKimball

closed 2019-06-28 by fpistm