stm32duino/Arduino_Core_STM32 #74: Ic2 slave SEND issue.

scargill, 2017-08-03:
I have the BLUEPILL board working - and took a look at the examples of I2c Slave..

The i2c Slave receive works - I can send several bytes to the slave (Bluepill) without issue.

Returning data however just isn't working.  The code I've used to test this works a treat on a nano and is nothing more than your example (but with send and receive functions in the same program).

I have PC13 flashing in the background (timeout - not delay) so I know things are working.  If I tell the Bluepill board to return 5 bytes and send a string using iWire.write("Hello ");  I don't get that back - and after several attempts at getting that returned data - the board locks up (the light stays on).
I tried your slave_send example - merely changing the address to 4.

A scan showed as I would expect, device 4 to be there - but sending a read command - nothing returned just as in my own code. I even put in a Serial1.writeln in the function to see if it was actually triggering... as I have on the read function... which reads a string sent to the board no problem - but when I2c asks the board for data, the requestEvent function is not even being called.

scargill, 2017-08-03:
The same code incidentally works just fine on an Arduino.... tested in the same way (but using Serial.println of course)

fpistm, 2017-08-04:
Hi @scargill,
As discussed on your other open issue, this is related to the Arduino_Core_STM32F1.

scargill, 2017-08-04:
Hi there - sorry - didn't quite follow the bit about the other open issue on this subject...   is this a bug, or am I doing something wrong?   I'm at a dead end here as I can get nothing out of the requestEvent function...

fpistm, 2017-08-04:
Please have a look at this [issue](https://github.com/stm32duino/Arduino_Core_STM32F1/issues/47)

closed 2017-08-04 by fpistm

scargill, 2017-08-04:
I did... and the mods to the library and the code to make simultaneous slave read/write work - and despite my best efforts - they simply don't work.  Will back off here until this is all official and working.