stm32duino/Arduino_Core_STM32 #75: Analog write

scargill, 2017-08-04:
Am I doing something wrong? Bluepill board..... 

analogWrite(PA1,x).  If I fire out 255 the output comes on - if 0, I get 0 but I'm not getting anything on other values?

According to the sheet in front of me - PA1 should be a PWM output.  I set it as as output....

fpistm, 2017-08-04:
@scargill 
Please do not enter issue related to BluePill on this repo while it is not officially release.
Submit your issue here:
[Arduino_Core_STM32F1/issues](https://github.com/stm32duino/Arduino_Core_STM32F1/issues)

fpistm, 2017-08-04:
To answer your question:
After checking the pin mapping on the F1 repo:
PA1 is:
`  { PA1,   GPIO_PIN_1,  GPIOA, GPIO_PIN_IO|GPIO_PIN_ADC,                     false },`
So there is no PWM on this pin nor DAC. So it is only a digital Pin (LOW or HIGH levels)
So writing:
 - 0 to 127 -> LOW
 - 128 to 255 -> HIGH

closed 2017-08-04 by fpistm