stm32duino/Arduino_Core_STM32 #76: Support for Nucleo F302R8 and STM32F072B-DISCO

bmitov, 2017-08-04:
Are there plans to support the Nucleo F302R8 and Discovery STM32F072B-DISCO with the Arduino Core?
I have been experimenting with both boards, but have not been able to get them to work  wit it :-( .

fpistm, 2017-08-04:
Hi @bmitov,
Have you take a look at [the wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))

bmitov, 2017-08-04:
Thank you! I am at least trying to get a Blink to work. I am working to add support for all the ST boards to Visuino - [https://www.visuino.com](https://www.visuino.com). I work with a lot of controllers, and add new boards and controllers every day. This means that I have very limited time that I can spend on specific board. If it does not have at least Arduino IDE support, I move to another board, as I have already more than 30 boards waiting for support :-( . The board also has to have a ready to use support as once I add it to Visuino, the Visuino users should be able to install and program the boards in their IDEs . I can't have each of them to manually configure it, as most of them are with absolutely no experience in programming or code.
I am trying to also organize classes for kids to program microcontrollers, and the people I am working with have a bunch of Nucleo, and Discovery boards. I was thinking that we can try to use them instead of Arduino to show the kids how easy it is to program the boards, and that is why I focused on this boards, but it seems that the Nucleo support is not in that good shape yet, so it is probably best to go with traditional Arduinos instead.

fpistm, 2017-08-04:
There is a lot of STM32 MCU based boards (Currently 136 MCU referenced). Adding a board variant could be done quickly (less than an hour when process is well understood ;)). Currently I'm working on several features which request lot of my time. You could create as many variants you want and submit PR then we could review it or help. There is several commits example in the repo.

bmitov, 2017-08-04:
I am already overloaded with other development. I would love to help, but I have never done this work, and I am completely out of resources :-( . I am simply trying to help promote the ST boards. I have to work on adding support of thousands of other modules, and boards in Visuino so there is really no time for this if the board is not supported already. Will have to wait I guess...

bmitov, 2017-08-04:
BTW: If you are interested in promoting specific reference design, and have it supported in Visuino, please let me know, and send me a board so I can add the design ;-) . I am trying to help microcontroller manufacturers to make their controllers popular among makers and people with no programming experience, and I would love to help you promote the ST boards. I already added support for the Maple Mini, and hope to add more boards.

bmitov, 2017-08-05:
Some update. Yesterday I exchanged the 2 boards for F303RE, and F401RE. It was late, but when I got back, I quickly ran a Blink test on the F303RE and it worked :-) . So I will be adding Visuino support for these 2 boards first, and as time permit will try to work on others, and may even try to do some mapping once I am more familiar with the ST boards and the firmware :-) .

fpistm, 2017-08-06:
Nice. Do not hesitate, if you have any questions

fpistm, 2017-11-07:
FY: Nucleo F302R8 support has been done by a contributor thanks a PR: #145 

I close this one.

closed 2017-11-07 by fpistm