stm32duino/Arduino_Core_STM32 #81: Nucleo F401RE pin mapping issue

gus-ghielec, 2017-08-15:
I have an LED connected to D34(PC6). My code works when I use D34 just fine but then it doesn't work if I use PC6 instead of D34. Both of these should be the same thing. There seem to be an issue when using the processor pin names.

This is necessary for those building products based on the nucleo board and want to use the processor pin name instead of the "Arduino" Dxx pin names.

fpistm, 2017-08-16:
Hi @gus-ghielec,
PinName and pin numbering is different. Arduino API use the pin numbering (Dx). When using the PinName (Pxy) its value is not equal to the pin number.
To use the PinName instead of  pin numbering use:
`uint32_t pinNametoDigitalPin(PinName p);`

So, `digitalWrite(D34, HIGH);` and `digitalWrite(pinNametoDigitalPin(PC6), HIGH);` are the same.
but `digitalWrite(pinNametoDigitalPin(PC6), HIGH);` not

rogerclarkmelbourne, 2017-08-16:
@fpistm 
I think you mean

digitalWrite(PC6, HIGH); not

BTW. I was not aware that the direct mapping does not work using the pin names like it does in libmaple and STM32Generic

fpistm, 2017-08-16:
Thanks @rogerclarkmelbourne 
Yes, not the same pinmapping. It is based on mbed pinmap.

rogerclarkmelbourne, 2017-08-16:
If someone made their own variant file, do you think they could map as 1:1 ?

i.e I know @gus-ghielec sells a custom board, and I think it may have pin names rather than pin numbers

Adminius, 2017-08-16:
it would be great if we can use Pxy names without pinNametoDigitalPin

fpistm, 2017-08-16:
No @rogerclarkmelbourne  as PinName is not an index.
[PinNames.h#L33](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/PinNames.h#L33)
It encodes the GPIO port and pin number on 8bits

rogerclarkmelbourne, 2017-08-16:
Yikes

This is going to be a rather confusing for anyone using the Discovery series boards, which are labelled with Pxn format names

RickKimball, 2017-08-16:
The pinname issue is kind of embedded deeply in these cores at this point.

fpistm, 2017-08-16:
> @RickKimball  The pinname issue is kind of embedded deeply in these cores at this point.
I will not change the pinmap management but I could grant this request. I will do a PR before the end of the week. I hope ;)

rogerclarkmelbourne, 2017-08-17:
I presume the problem is all associated with the Analog pin numbers needing to be Arduino compatible :-(

fpistm, 2017-08-17:
Yes, always the same problem... to be or not to be Arduino compatible. :')

rogerclarkmelbourne, 2017-08-17:
Frederic

I'm not sure if its something you can decide...

But given the choice of either being able to use the real Port / Pin name or supporting the Arduino analogue pin number system, I would use Port / Pin 

There are so many other things which are not going to be AVR compatible, that trying to make the Analog pin names compatible will be a minor change compared with the Harvard vs Von Neumann architecture and the size of int and float etc, that people will have to get used to, if migrating from AVR.

Adminius, 2017-08-17:
Pxy/Dx and Ax, deal? :)

fpistm, 2017-08-17:
I will only add possibility to use pin name as an index matching the Dx. (no loop ;))
So Analog will be the same.

PYn == Dx == x for digital
PYn == Ax == x == Dy for analogRead
PYn == Ax == Dy for analogWrite

fpistm, 2017-08-17:
(for Nucleo F429ZI):

**Digital:**
```
digitalWrite(33, HIGH); 
digitalWrite(D33, HIGH);
digitalWrite(PB0, HIGH);
digitalWrite(pinNametoDigitalPin(PB_0), HIGH);
```
**AnalogRead:**
```
analogRead(A0);
analogRead(0);
analogRead(PA3);
analogRead(pinNametoDigitalPin(PA_3));
analogRead(D78);
analogRead(78);
```

**AnalogWrite:**
```
analogWrite(A0);
analogWrite(PA3);
analogWrite(pinNametoDigitalPin(PA_3));
analogWrite(D78);
analogWrite(78);
```
`analogWrite(0);` write on D0 (PG9)

rogerclarkmelbourne, 2017-08-17:
Excellent.

fpistm, 2017-08-18:
PR #83 merged

closed 2017-08-18 by fpistm