stm32duino/Arduino_Core_STM32 #93: I2C pins on STM (Nucleo STM32F411RET6 - D14/D15)

valoni, 2017-08-26:
Tried some examples for I2C pins - but those do not work or need different way to approach this problem
 (i mean still have problem or need to fix something to be able use of I2C)

**I2C Scanner example do not work** 
https://playground.arduino.cc/Main/I2cScanner  

**and Library to use with (i2c lcd1602 - with arduino work with no problem with Nucleo NOT)**
https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library  not work

valoni, 2017-08-26:
**resolved this and both libraries working well** 
way of resolving was "this one is not documented well"

with two 4k7 pull-up resisters on between Vcc(5v)-SCL and Vcc-SDA pins on LCD1602

closed 2017-08-26 by valoni

fpistm, 2017-08-27:
Hi,
One other way could be to set `GPIO_PULLUP` instead of GPIO_NOPULL
[NUCLEO_F411RE/PeripheralPins.c#L75](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_F411RE/PeripheralPins.c#L75)

valoni, 2017-08-27:
If i change as you say then it will act in original way as arduino without additional resistor for pullup since it will have same state
thanx a lot for info
Just a some question 
1.can I use arduino ethernet as is or need to modify any pin headers like this way.
2.And how to declare more serial in nucleo
Example in Mega/Due it was Serial,Serial1.Serial2
(Or i need to use software serial for more serial ports)
So there is good to have more info/documentation about those kinds
Regards,Valoni Sunday,

fpistm, 2017-08-27:
For Ethernet, as F411 have the Arduino Uno connector it should be ok. But you will have to connect the ICSP manually to the D11/D12/D13 pin (SPI signals) as the Nucleo do not have the ICSP connector.

For Serial, enable it thanks the menu Other Serial.

It's planned to add some section in the wiki. Libraries, API,...

valoni, 2020-03-04:
Hey,

I just signed the petition "Qeveria e Kosovës: Qe të mos shkarkohet
Ambasadorja Vlora Çitaku" and wanted to see if you could help by adding
your name.

Our goal is to reach 100 signatures and we need more support. You can read
more and sign the petition here:

http://chng.it/g9gTzBYSgD

Thanks!
Valon