stm32duino/Arduino_Core_STM32 #104: Timer driven interrupts

Adminius, 2017-09-11:
Hey,
is something like this 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/A_STM32_Examples/examples/Maple/TimerInterrupts/TimerInterrupts.ino#L35
planned for this core?

fpistm, 2017-09-11:
Hi @Adminius,
it should be fine. 
I though to extend Timer but I do not know when.
Any contribution are welcome. ;)
Currently next major steps planned for this core are the Low power management for Lx series and ADC modes which will be extended.

Adminius, 2017-09-11:
Ok thx, @fpistm . i can wait. I will also try to adopt it, but i'm not sure whether I can manage it :( 
Low power sounds good 👍, but BluePill support is more impotant (at least for me)...

fpistm, 2017-09-11:
I know I'm always working on BP integration...

fpistm, 2017-11-07:
Will be covered by #146

closed 2017-11-07 by fpistm