stm32duino/Arduino_Core_STM32 #107: Question about LED_BUILTIN

ekawahyu, 2017-09-18:
Hi, I am trying to port STM32F072 Discovery and I know that it has 4 LEDs. My question is related to these lines within variant.h header file:

//On-board LED pin number
#define LED_BUILTIN             Dx
#define LED_GREEN               LED_BUILTIN

If I considered all of those 4 LEDs on STM32f072 Discovery are not LED_BUILTIN and I don't want to have one, how should I modify the #define LED_BUILTIN value? Would an empty value work just fine?

fpistm, 2017-09-18:
Hi @ekawahyu 
This define value is only for convenience.
If you do not want define it the official arduino blink sketch will not build as it use it but it's up to you or you can define it on other dedicated pin for a LED

For example, for the Nucleo F429zi which also have 4 on board leds, it is defined like this for convenience
```
//On-board LED pin number
#define LED_BUILTIN             33
#define LED_GREEN               LED_BUILTIN
#define LED_BLUE                73
#define LED_RED                 74
```

LMESTM, 2017-09-18:
I also agree that it is best to consider that one of the 4 LEDS is a built-in (default) one.

ekawahyu, 2017-09-18:
Got it! Is there a way to ask questions (like a forum) instead of raising an issue on GitHub? ST Forum or STM32Duino one maybe? You guys can close this issue. Thank you.

fpistm, 2017-09-18:
Yes on stm32duino.com forum.
As mentionned in the [readme.md](https://github.com/stm32duino/Arduino_Core_STM32#troubleshooting)

closed 2017-09-18 by fpistm