stm32duino/Arduino_Core_STM32 #108: Issue to build with 'Aggressively cache compiled core' enabled on Windows

fpistm, 2017-09-20:
Depending of the board and menu option after one build it is not possible to build again when  _Aggressively cache compiled core_ is enabled in the Arduino IDE references.

Example
1.  First build is ok:
`Archiving built core (caching) in: C:\Users\guest\AppData\Local\Temp\arduino_cache_703236\core\core_stm_stm32_Disco_board_part_num_DISCO_L475VG_IOT,upload_method_MassStorageMethod,Other_serial_enable_Serial,USB_interface_enable_USB,opt_oslto_4e2f2f4828dcde7e63516bf59d4ffb2d.a`
2. Second one failed:
`arm-none-eabi-gcc: error: C:\Users\guest\AppData\Local\Temp\arduino_build_927530/..\arduino_cache_703236\core\core_stm_stm32_Disco_board_part_num_DISCO_L475VG_IOT,upload_method_MassStorageMethod,Other_serial_enable_Serial,USB_interface_enable_USB,opt_oslto_4e2f2f4828dcde7e63516bf59d4ffb2d.a: No such file or directory`

The path is higher to 260 (MAX_PATH of windows) that's why there is an error. Even if this limitation is disabled it seems Arduino is not able to deal with.
Issue met with Win7 and Win10.

One solution is to shorten option in board.txt else disable the _Aggressively cache compiled core_ in the Arduino IDE references.

fpistm, 2017-09-20:
With #109 
`C:\Users\guest\AppData\Local\Temp\arduino_build_927530/..\arduino_cache_985791\core\core_stm_stm32_Disco_pnum_DISCO_L475VG_IOT,upload_method_MassStorage,xserial_none,usb_none,opt_oslto_4e2f2f4828dcde7e63516bf59d4ffb2d.a`
Now 220 instead of 266

fprwi6labs, 2017-09-21:
Did you open an issue to Arduino?

fpistm, 2017-09-21:
Not yet but I will do it.
however, I think it is good to shorten some options. Moreover, it is more explicit name.

fprwi6labs, 2017-09-21:
Of course it is better. So now we must be careful when we choose an option name. Sneaky bug.

closed 2017-09-21 by fpistm

fpistm, 2017-09-24:
Issue opened: [Arduino#6757](https://github.com/arduino/Arduino/issues/6757)

PR #109 (2017-09-21): Shorten options names to avoid issue with aggressively cache compiled core
Fix #108
commit 8eaaf4b7 (2017-09-21): Shorten options names to avoid issue with aggressively cache compiled core

Fix #108