stm32duino/Arduino_Core_STM32 #111: TIMER_TONE with value of TIM14 or TIM15 do not seem to work

ekawahyu, 2017-09-23:
I am porting the core to work with STM32F072RB Discovery Kit. Everything seems to be working just fine, except that tone() does not generate any sound with TIM14 or TIM15. I haven't done any test with all of timers, but a quick test with TIM3 seemed to generate some tone.

Related thread:
http://www.stm32duino.com/viewtopic.php?f=49&t=2614&p=34733#p34733

fpistm, 2017-09-25:
Reproduced on Nucleo-F091RC.
@fprwi6labs  any clues on this issue?

fprwi6labs, 2017-09-25:
Reproduced too on Nucleo-F091RC. TIM1 failed too. 
IRQ handler for timer 15,16,17, ... are missing. But it do not explain why TIM14 or TIM1 don't work.

fprwi6labs, 2017-09-25:
@fpistm the IRQHandler names are wrong for F0. I will fix them.

fpistm, 2017-09-25:
Thanks @fprwi6labs 
I've forgot to add those IRQ handler and update name which is different per series...

fpistm, 2017-09-27:
#112 merged

closed 2017-09-27 by fpistm