stm32duino/Arduino_Core_STM32 #113: Serial3 broken, fix created

mahesh2000, 2017-09-25:
Serial3, which used to work back in 2016, is now broken. Fix is below:

..\STM32\hardware\stm32\2017.9.22\boards.txt
..\STM32\hardware\stm32\2017.9.22\variants\NUCLEO_L476RG\variant.h
..\STM32\hardware\stm32\2017.9.22\variants\NUCLEO_L476RG\variant.cpp

Cleanup in #defines for Serial3 is needed.

Also, in ..\STM32\hardware\stm32\2017.9.22\variants\NUCLEO_L476RG\PeripheralPins.c,
const PinMap PinMap_UART_TX[] has UART4 mapped to both PA_0 and PC_10. is this  correct? the same with PA_1 and PC_11.

[NUCLEO_L476RG.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/1330136/NUCLEO_L476RG.zip)

fpistm, 2017-09-26:
Hi @mahesh2000,

I think you talk about:
https://github.com/stm32duino/Arduino_Core_STM32L4/blob/d30e65eed475fcdcafb9ce26e74be1055af860a5/variants/STM32L476RG_Nucleo/variant.cpp#L145

It is not really broken. Serial3  (map to UART4) is only not defined by default in this core.
For the PeriperalPins.c this is correct. You can use one or the other pin to the Serial connection.
Ex:
```
HardwareSerial Serial3(PA_1, PA_0);
HardwareSerial Serial3(PC_11, PA_0);
HardwareSerial Serial3(PA_1, PC_10);
HardwareSerial Serial3(PC_11, PC_10);
```
It's up to you to connect the serial interface on the right pins.

Howerver, I think you raised one issue:
How to define an extra Serial and process it to the SerialEventRun().

closed 2017-10-06 by fpistm

mahesh2000, 2017-10-06:
hi, afaik, hardware serial ports numbers on the  STM32L476 Nucleo-64 are specified to certain pins -- it would make sense to have the same nomenclature in the variant.cpp rather than let the user define his/her own. What do you think?

fpistm, 2017-10-06:
Hi, in fact since #122 is merged, there is no more need to define HardwareSerial in the variant.cpp.
Serial are now handle more generically.
User could (re)define them from the sketch depending of his need (even the SerialEvent)
If you want Serial3 enable the 'First third option in the Serial instance menu, this will enable the Serial 3.

fpistm, 2017-10-06:
By default, when Serial 3 is enabled on Nucleo L476RG pins will be PB10(Tx) PB11(Rx).
To change them call in the sketch:
```
  Serial.setRx(PC5);
  Serial.setTx(PC4);
```
before the `Serial3.begin(9600)`

PR #122 (2017-10-06): Make Serialx instance generic
Previously Serialx instance was defined by the variant.
It was not enough generic and flexible.
    
With this patch, all available Serialx instance are handled dynamically.
Where 'x' is the U(S)ART number.
Currently, STM32 MCU could have up to 10 U(S)ART.
    
Serial is the generic name used in Arduino sketches.
Serial is now mapped to the Serialx instance linked to the com port in the variant.h (mainly one linked to ST-Link).
To define the generic Serial, define this in variant.h:
```
// Define here Serial instance number to map on Serial generic name
#define SE [... cut]