stm32duino/Arduino_Core_STM32 #117: Nucleo-F303K8 Support

tofuman0, 2017-09-28:
I've been trying to get the Nucleo-F303K8 working using the Arduino IDE. I've attempted to create a variant. The project compiles and uploads but the F303K8 doesn't appear to do anything. I'm a complete newbie so it'll most likely be something I've done wrong. I've attached the variant I've made. Some guidance would be greatly appreciated.
[NUCLEO_F303K8.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/1342326/NUCLEO_F303K8.zip)

Thanks.

rogerclarkmelbourne, 2017-09-29:
For a board to appear in the Boards Menu, it needs to be included in the boards.txt file

e.g. Look at this section and duplicate it

https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L212-L248

And change all references to your board, including the reference Id at the start of each line, the menu item for your one and only board in this section, the publically visible name, the variant folder etc etc etc

Once you have updated boards.txt, you need to restart the IDE (you need to do this every time you make a change to boards.txt)

PS. I think there is a way to add a 'Board' using the Boards manager JSON packaging system, but you should get things working by manually editing boards.txt first, and then work towards building the complex JSON file that is required to install via the Boards manager.

Or when its working, you may like to submit a PR for its inclusion

fpistm, 2017-09-29:
Have you take a look here:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)#9---declare-the-variant

fpistm, 2017-09-29:
I've tested quickly,
By adding in boards.txt in Nucleo_32 section:
```
# NUCLEO_F303K8 board
# Support: Serial1 (USART2 on PA3, PA2)
Nucleo_32.menu.pnum.NUCLEO_F303K8=Nucleo F303K8
Nucleo_32.menu.pnum.NUCLEO_F303K8.node=NODE_F303K8
Nucleo_32.menu.pnum.NUCLEO_F303K8.upload.maximum_size=65536
Nucleo_32.menu.pnum.NUCLEO_F303K8.upload.maximum_data_size=12288
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.mcu=cortex-m4
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.f_cpu=72000000L
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.board=NUCLEO_F303K8
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.series=STM32F3xx
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.product_line=STM32F303x8
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.variant=NUCLEO_F303K8
Nucleo_32.menu.pnum.NUCLEO_F303K8.build.cmsis_lib_gcc=arm_cortexM4l_math
```
Then Blink test was not working. SystemClock_Config seems not correct.
So, I've replaced by one from the STM32CubeF3 example for Nucleo D303K8:
```
WEAK void SystemClock_Config(void)
{

  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_OscInitTypeDef RCC_OscInitStruct;
  
  /* HSI Oscillator already ON after system reset, activate PLL with HSI as source */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_NONE;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL16;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }

  /* Select PLL as system clock source and configure the HCLK, PCLK1 and PCLK2 
     clocks dividers */
  RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2);
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;  
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }
  
    /**Configure the Systick interrupt time 
    */
  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);

    /**Configure the Systick 
    */
  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}
```

Then Blink is ok.

tofuman0, 2017-09-29:
I should have also mentioned I had updated the boards.txt. @fpistm Brilliant. I'll give this a go later today. Thank you for your help.

tofuman0, 2017-09-29:
@fpistm It seems to work pretty well. I had to make some further changes to my pins to get i2c working but other than that it worked. However I2C is extremely slow. I'm using a SSD1306 and if I output a counter in a loop with no delays it increments 2 every second. Is there anyway to speed this up?

I also tried to get CubeMX to generate the same SystemClock_Config but no matter how I configured the RCC and clocks it generates with RCC_OSCILLATORTYPE_HSI instead of RCC_OSCILLATORTYPE_NONE.

Thanks.

fpistm, 2017-09-30:
You could try to change the I2C freq, thanks: `Wire.setClock(I2C_400KHz)` by defaut it's I2C_100KHz
About the SystemClock_Config, I only use one from the cube. It's not the best one I used.
I will try to create one to have the best freq.

tofuman0, 2017-09-30:
Wire.setClock(I2C_400KHz) Doesn't appear to make any difference.

Thanks.

fpistm, 2017-10-02:
@tofuman0, please, could you share your updated variant and your sketch

tofuman0, 2017-10-02:
@fpistm sure. Here it is:

#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);
uint32_t counter = 0;

void setup() {
  // put your setup code here, to run once:
  u8g2.begin();
  Wire.begin();
  u8g2.setFont(u8g2_font_calibration_gothic_nbp_tr);
}

void loop() {
  // put your main code here, to run repeatedly:
  char tempStr[20];
  counter++;
  sprintf(tempStr, "%d", counter);
  u8g2.clearBuffer();
  u8g2.drawStr(64, 31, tempStr);
  u8g2.sendBuffer();  
}

I've benchmarked the Nucleo-F303K8 and it is running as expected its only i2c that causes it to run extremely slow.

fpistm, 2017-10-02:
Thanks.
it is really needed to call  `Wire.begin();`?
It should be performed in the ` u8g2.begin();`

Could you provide also the zip of the variant you used  or an url where I can find it (github?)?

tofuman0, 2017-10-03:
[NUCLEO_F303K8.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/1353979/NUCLEO_F303K8.zip)
Here is the variant I'm using.

True I could remove Wire.begin().

Thanks.

fpistm, 2017-10-05:
Hi @tofuman0,
I've tested your variant.
I've updated your variant based on the [UM1956 STM32 Nucleo-32 board](http://www.st.com/resource/en/user_manual/dm00231744.pdf):
    - Remove Serial2 as only 2 USART available.
    - Update SystemClock_Config to configure at 64 Mhz using HSI.
      By default, there is no HSE (X1) on this board.
      Configure RCC_PERIPHCLK_I2C1|RCC_PERIPHCLK_ADC12;
    - By defaut, SB16 and SB18 are ON. So, A4 and A5 are connected
      to PB7/PB6 instead of PA5/PA6 which must be configured as
      input floating.
      See Table 7. Solder bridges of [UM1956 STM32 Nucleo-32 board](http://www.st.com/resource/en/user_manual/dm00231744.pdf)

I've tested SPI and also I2C which is now more accurate.
You could find the updated version on the branch Nucleo_F303K8 available here:
[fpistm/Arduino_Core_STM32/tree/Nucleo_F303K8](https://github.com/fpistm/Arduino_Core_STM32/tree/Nucleo_F303K8)
3 commits:

1. Clean up variant and updated config: https://github.com/fpistm/Arduino_Core_STM32/commit/2080bf01e415766042026630db0a72177ad6711f
2. Add entry in board.txt: https://github.com/fpistm/Arduino_Core_STM32/commit/744d86a542db11419281d95cdbe7e678316f4c2e
3. Added variant for Nucleo-F303K8 from @tofuman0: https://github.com/fpistm/Arduino_Core_STM32/commit/8d183b91b0a25a3838982f63453aca80de1d5ca1

You should have already commit 2 and 3. So, commit 1 is the main merge on your code.
Let me know if it is ok

tofuman0, 2017-10-05:
Thanks. Just tested it and it much faster now. I added timing code to the code and ran it on the Arduino Nano, Teensy 3.2 and Nucleo32-F303K8 and the following results displaying 0-1000

Arduino Nano - 45.000s
Nucleo32-F303K8 - 33.837s
Teensy - 36.167s

fpistm, 2017-10-06:
Nice.
So, all is ok now?
I could add the variant in the core if you want?
Could you give me your mail linked to yout github account then I will add  your signed off for the variant as your are the main author  :)

fpistm, 2017-10-09:
PR #127 submitted

tofuman0, 2017-10-09:
Sorry. I've been away for a few days, but can see you've added the variant. Did you need me to do anything?

Thanks.

fpistm, 2017-10-10:
No worries.
If it is ok for you I will merged it.
If you want I could add your email address in the signed off if you give me one.

fpistm, 2017-10-12:
I've merged it.
Thanks for your work.

closed 2017-10-12 by fpistm