stm32duino/Arduino_Core_STM32 #126: Build error with Firmata

fpistm, 2017-10-09:
Since the clean of header files including Firmata do not build.
Still one issue with multiple inclusions with pins_arduino.h.
```
/local/data/work/stm32/stm32duino/arduino-1.8.4/portable/sketchbook/libraries/Firmata/Boards.h:790:72: error: 'pinIsSerial' was not declared in this scope
 #define IS_PIN_DIGITAL(p)       (digitalPinIsValid(p) && !pinIsSerial(p))
```
Fixed on going.

fpistm, 2017-10-10:
Fixed in #128

closed 2017-10-10 by fpistm

xzsfd2457, 2019-05-06:
I am facing a problem while using Softserial header file for STM32F401RE microcontroller, then it tells the Softserial directory is not available.

fpistm, 2019-05-06:
Hi @xzsfd2457,
Please open a new issue and describe your issue.
environnement (host OS, Arduino IDE version, core version,...), sketch,...
Thanks in advance