stm32duino/Arduino_Core_STM32 #132: UIPEthernet library - doesnt work

valoni, 2017-10-16:
i've got this kind of error 

../Enc28J60Network.h:106:5: error: #error "Not defined ENC28J60_CONTROL_CS!"
../Enc28J60Network.h:131:5: error: #error "Not defined SPI_MOSI!"
../Enc28J60Network.h:156:5: error: #error "Not defined SPI_MISO!"
../Enc28J60Network.h:180:5: error: #error "Not defined SPI_SCK!"

with default examples

fpistm, 2017-10-16:
Hi @valoni,

sounds normal as this library do not handle this core.
It seems it's should be easy to update by adding under ARDUINO_ARCH_STM32 the right pins assignment in Enc28J60Network.h using the  default Arduino naming (SS, MOSI, MISO, SCK).

fpistm, 2017-10-16:
Could you try with this patch.
Build is ok but I do not make any test with a board.
[STM32.patch.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/1387253/STM32.patch.txt)

valoni, 2017-10-16:
with this patch i can compile now .

so now i'm going to test it to see did work it

valoni, 2017-10-16:
just a questions why this patch is needed to compile under STM32F411ret6 Nucleo which seems was already defined

fpistm, 2017-10-17:
In fact,  Arduino_Core_STM32 core is a generic one for STM32 MCU. Only the ARDUINO_ARCH_STM32 is needed.
The other switches you've mentioned are for another STM32 core.

valoni, 2017-10-17:
how to use for example switch for Nucle STM32F411RET6 without applying Patch !!!!

because without patch i've got errors as presented (if apply patch it compile)

../Enc28J60Network.h:106:5: error: #error "Not defined ENC28J60_CONTROL_CS!"
../Enc28J60Network.h:131:5: error: #error "Not defined SPI_MOSI!"
../Enc28J60Network.h:156:5: error: #error "Not defined SPI_MISO!"
../Enc28J60Network.h:180:5: error: #error "Not defined SPI_SCK!"

fpistm, 2017-10-17:
The `TARGET_NUCLEO_F411RE` is for mbed so you can't use it for Arduino.
For this one, fo r example:
`#if defined(STM32F3)                //This is workaround for stm32duino STM32F3`
it is for the @rogerclarkmelbourne core:
https://github.com/rogerclarkmelbourne/Arduino_STM32

valoni, 2017-10-18:
ok i understand n ow so the patch is and univesal approach for STM32 MCU (F1,F2,F4,...)
so independet of mcu it allow used in same way for all those   ARDUINO_ARCH_STM32 (on thise arch is all Nucleo64 board which are appeared on Arduino IDE Menu - if i am not wrong)

fpistm, 2017-10-18:
Yes, this core (Arduino_Core_STM32) appears in the menu like this:
[image]
[image]

Note, this is old screenshot they are more boards available now.
See the Wiki: https://github.com/stm32duino/wiki/wiki/Getting-Started

fpistm, 2017-10-24:
@valoni ,
have you tested with the library patched with the hardware.
If it is ok I will made a PR.

valoni, 2017-10-27:
i'm abrooad and today i will return at home 
so this weekend i will test it with hardware and will inform about it

fpistm, 2017-10-27:
Ok. Thanks @valoni

valoni, 2017-10-29:
**i can compile after applying patch**
but tests with two differents HARDWARE do not work and with this sample below 
**(i tested and default samples too but not work)**

> /*  Web Server */
> #include <UIPEthernet.h>
> 
> byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
> byte ip[]={192, 168, 1, 177};
> byte subnet[]={255, 255, 255, 0};
> byte gateway[]={192, 168, 1, 1};
> byte mydns[]={192, 168, 1, 1};
> 
> // (port 80 is default for HTTP):
> EthernetServer server(80);
> 
> void setup() {
>   // Open serial communications and wait for port to open:
>   Serial.begin(9600);
> 
>   // start the Ethernet connection and the server:
>   Ethernet.begin(mac, ip, mydns, gateway, subnet); 
>  
>   server.begin();
>   Serial.print("server is at ");
>   Serial.println(Ethernet.localIP());
> }
> 
> 
> void loop() {
>   // listen for incoming clients
>   EthernetClient client = server.available();
>   if (client) {
>     Serial.println("new client");
>     // an http request ends with a blank line
>     boolean currentLineIsBlank = true;
>     while (client.connected()) {
>       if (client.available()) {
>         char c = client.read();
>         Serial.write(c);
>         // if you've gotten to the end of the line (received a newline
>         // character) and the line is blank, the http request has ended,
>         // so you can send a reply
>         if (c == '\n' && currentLineIsBlank) {
>           // send a standard http response header
>           client.println("HTTP/1.1 200 OK");
>           client.println("Content-Type: text/html");
>           client.println("Connection: close");  // the connection will be closed after completion of the response
>           client.println("Refresh: 5");  // refresh the page automatically every 5 sec
>           client.println();
>           client.println("<!DOCTYPE HTML>");
>           client.println("<html>");
>           // output the value of each analog input pin
>           for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
>             int sensorReading = analogRead(analogChannel);
>             client.print("analog input ");
>             client.print(analogChannel);
>             client.print(" is ");
>             client.print(sensorReading);
>             client.println("<br />");
>           }
>           client.println("</html>");
>           break;
>         }
>         if (c == '\n') {
>           // you're starting a new line
>           currentLineIsBlank = true;
>         } else if (c != '\r') {
>           // you've gotten a character on the current line
>           currentLineIsBlank = false;
[... 210 lines cut ...]
                    sendHTTP(client, httpHeader, movedPermanently(1));
                    continue;
                }
 
                if(cmd == -1) {
                    httpHeader = UNAUTHORIZED;
                    httpContent = "<h1>401 Unauthorized</h1>";
                    sendHTTP(client, httpHeader, httpContent);
                    continue;
                }
 
                if(cmd == ON) {
                    sw = ON;    // turn the switch on
                }
 
                if(cmd == OFF) {
                    sw = OFF;   // turn the switch off
                }
 
                httpHeader = HTTP_OK;
                sendHTTP(client, httpHeader, showWebPage(sw));
            }
        }
    }
}

fpistm, 2017-10-29:
Hi @valoni 
unfortunately I do not have this hardware.
For the wrong IP adress print have you check on your box the real IP address?
It seems the [UIPEthernetClass::localIP()](https://github.com/UIPEthernet/UIPEthernet/blob/602a12fa317eae911482de89e648dcef608c047f/UIPEthernet.cpp#L158) uses some conversion which maybe could cause display issue.
Probably some other update are required but without the hardware I could not investigate.

valoni, 2017-10-29:
I set ip statically

On pc i set:
192.168.1.175 ip
255.255.255.0 sub
192.168.1.1 gtw
192.168.1.1 dns

On nucleo (see code attached above)
192.168.1.177
255.255.255.0
192.168.1.1
192.168.1.1

Tried to ping from pc (nothing)

So for this reason  i provided 2 different code for different platforms (which i tested with NucleoSTM32F411Ret6)

1.Arduino IDE (that not work)
2.mbed online compiler(that work pretty well)

romainreignier, 2017-11-01:
I have tried the patch from @fpistm. It allows to build the library but it does not work.

@valoni was right to point the faulty IPAddress, there is an endianness issue, replacing `LITTLE_ENDIAN` by `BIG_ENDIAN` in the following line made the demo sketch `EchoServer` works:

https://github.com/UIPEthernet/UIPEthernet/blob/602a12fa317eae911482de89e648dcef608c047f/utility/uip-conf.h#L125

Note that according to the other ports, I have tried to change the SPI clock divider here:

https://github.com/UIPEthernet/UIPEthernet/blob/602a12fa317eae911482de89e648dcef608c047f/utility/Enc28J60Network.cpp#L144

But with the default value on a L476 board, it works and the ping is faster.

fpistm, 2017-11-01:
Hi @romainreignier ,

thanks for the feedback.
Sounds strange for the endianness. Mainly linked to the HTONS:
https://github.com/UIPEthernet/UIPEthernet/blob/4966ee06bc29d42749ca3a3dde5d39cbe9183e98/utility/uip.h#L1066

By default SPI use a default clk of 4MHz and defined automatically the best clock divider.
So depending of the clock divider you tried maybe it runs slowly than 4MHz depending of the clk freq of the SPI instance.

romainreignier, 2017-11-01:
Yes exactly, it is the `HTONS` macro that made me change the endianness definition.
I have also found it weird because for the other ARM platforms, the endianness was not changed so I wondered if you have change anything in the Arduino toolchain to create that issue.

In fact the `HTONS` only do usual bit shifting so I do not know where the issue is.

Actually, the IP address seems the issue because even with the default settings, the MCU is communicating without issue with the ENC28J60 chip, I can see the traffic when I ping it using the logs on the serial port. But it does not respond because the address is faulty.

About the clock divider, it is true that I did not check the default value. I have measured the frequency with a logic analyzer for the 32, 16 and 8 dividers but because all were working I decided to let the default value without measuring it.

valoni, 2017-11-12:
on config file **uip-conf.h**

a applied condition as below too 

 * CPU byte order.
 *
 * \hideinitializer
 */
 
#if defined(ARDUINO_ARCH_STM32)
#define UIP_CONF_BYTE_ORDER      BIG_ENDIAN
#else
#define UIP_CONF_BYTE_ORDER      LITTLE_ENDIAN
#endif
 
/**
 * Logging on or off
 *

and now UIPETHERNET work well with Nucleo STM32F411RET6

fpistm, 2017-11-12:
Hi @valoni 
thanks for the test. Sorry, I do not have time to check in my side.
Anyway, it is strange to have to set as big endian, it should be fine to understand why.

valoni, 2017-11-12:
I belive for ethernet way should be BIG_ENDIAN
not LITTLE_ENDIAN maybe this was bug in ARDUINO so original author used manually way for

valoni, 2017-11-12:
..

valoni, 2017-11-20:
Another problem did anyone tested on Discovery STM32F407VGT6 - UIPETHERNET seems not work on this board even with patches above (maybe is problem with clock speed) 

but on the Nucleo Boards is tested and work for following boards ; 

STM32F411RET6 - YES (without patched clock)
(i belive on STM32F401 will work with no problem since it close to F411 board but with less speed and ram , i will test it on GHI Brainpad board it have STM32F401 to be sure)

STM32L476 - Yes (with and without patched clock)

fpistm, 2017-11-22:
@valoni I think the patch I provided to you as example probably needs some review.
Maybe you could submit an issue on the UIPEthernet github librariy to have some advices (required SPI speed,...).

valoni, 2017-11-23:
I mean first you need to post this patch "to make working" UIPETHERNET for now in 
Nucleo 401,407,L476 ... at least " - do not forget to patch also BIG/Little ENDIANESS problem too.

and after we go in another issue - what is working and what not working parts in this way 
we can optimise and find problems with STM32 CORE and going to eliminate it.

valoni, 2018-01-19:
similiar to this our ones and regards to authors 

on part  https://forum.pjrc.com/threads/47791-Teensy-3-2-ENC28J60-Strange-behavior?p=159635&viewfull=1#post159635   problem persist to the macros HTONS() and htons() as in this thread was from @romainreignier  specified exact same problem.

fpistm, 2018-01-21:
Hi guys,

I've submitted the support of this core to the UIPEthernet repo with a proper fix for the endianness issue I think.
https://github.com/UIPEthernet/UIPEthernet/pull/26

closed 2018-01-21 by fpistm