stm32duino/Arduino_Core_STM32 #144: Get old Version with Boardmanager

daUser, 2017-11-05:
Hi,
i would like to use Arduino_Core for my STM32_F303K8. So I added the Packages like in your Tutorial: https://github.com/stm32duino/wiki/wiki/Boards-Manager
But on this way I get an old version without the F303K8.

Can you fix this?

fpistm, 2017-11-05:
Hi @daUser,
The board manager provides the packaged versions of this repo when I do a release.
The Nucleo F303K8 will be available in the next release.
https://github.com/stm32duino/Arduino_Core_STM32/releases
I've planned to make one release in the 2 coming weeks.

daUser, 2017-11-05:
Hi @fpistm,
will it be possible, du mark the Boards in the Overview ( https://github.com/stm32duino/Arduino_Core_STM32 ) as "in next Release - use git repo" or so?
I get it working over the git repo, but it's a lite bit confusing to read that it works, download it how its descriptet in the Tutorial and then it doesn't work ;-)

fpistm, 2017-11-05:
Right. I think I will change the Readme.md by splitting in 2 subsections the boards section. 1 for the boards available in the official release and one for the next release.

fpistm, 2017-11-06:
I've updated the readme

closed 2017-11-06 by fpistm