stm32duino/Arduino_Core_STM32 #157: SPI: CS configuration "lockout", code, documentation, examples

gnbl, 2017-11-22:
I'm using SPI3 (seems to be the default for SPI.begin() ) on a Nucleo-L432KC (STM32L432KC) like so

```
#include <SPI.h>
SPIClass io(D11, D12, D13);//, A3);

io.beginTransaction( A3, SPISettings(12e6, MSBFIRST, SPI_MODE0) );
io.transfer16(A3, 0xF0F1 );
```
because I prefer explicit configuration and later also want to use SPI1.

**Configuration "lockout"**
[SPIClass must not instanciated with `ssel`](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SPI/src/SPI.h#L108) or [beginTransaction(uint8_t pin,...](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SPI/src/SPI.h#L148) does not configure the CS pin.
Could this not also prevent using SPI interfaces other than default?

**Code**
Maybe it is also possible to simplify the keeping of the CS state, imo variables , (private) members, parameters have very similar symbols (are they redundant?) and are even used mixed, which makes it a little hard to read:
```
pin
ssel
_pin
_spi.pin_ssel
_CSpin
```
(magic) constants:
```
-1
0xFF
CS_PIN_CONTROLLED_BY_USER
NC
```

**Documentation**
Why are there almost no comments on what's happening (I know: "because I did not write them")? But seriously. Good code may not need comments, but not comments don't necessarily make good code..

**Examples**
It would be nice to have an example or an explanation on how to use different SPI peripherals. From what I understand, the CS pin determines the hardware peripheral!?

fprwi6labs, 2017-11-24:
Hi @gnbl ,

Thank you for your feedback.
There is documentation in the [wiki](https://github.com/stm32duino/wiki/wiki/API#spi) about the SPI API.

fpistm, 2017-11-25:
Any feedback?

fprwi6labs, 2017-11-27:
I reopened because we must increase the quality of comments in this code.

closed 2017-11-30 by fpistm