stm32duino/Arduino_Core_STM32 #160: PROBLEM  void TIM6_DAC_IRQHandler(void)

franciscogimeno2000, 2017-11-29:
Hello friends, I have this code to use timer 6:

void tim6_init() {
	RCC->APB1ENR |= RCC_APB1ENR_TIM6EN;                 // Enable TIM6 clock
	TIM6->PSC = 41999;                                  // Set prescaler to 41999
	TIM6->ARR = 9000;                                   // Set auto-reload to 5999
	TIM6->CR1 |= TIM_CR1_OPM;                           // One pulse mode	
	TIM6->EGR |= TIM_EGR_UG;                            // Force update
	TIM6->SR &= ~TIM_SR_UIF;                            // Clear the update flag
	TIM6->DIER |= TIM_DIER_UIE;                         // Enable interrupt on update event
	NVIC_EnableIRQ(TIM6_DAC_IRQn);                      // Enable TIM6 IRQ
	TIM6->CR1 |= TIM_CR1_CEN;    
}

void TIM6_DAC_IRQHandler(void)
{
digitalWrite(8, HIGH); //test only

if (TIM6->SR & TIM_SR_UIF != 0) // If update flag is set
{
TIM6->SR &= ~TIM_SR_UIF;    // Interrupt has been handled
digitalWrite(8, HIGH); //test only
}
}

The code is compiled well and the counter counts according to plan.

The problem is that when the end of the account reaches the micro is blocked and never enters the
interruption, it's as if he were going to look for her and he would not find her.

Any suggestions?

fpistm, 2017-11-29:
Hi @franciscogimeno2000 ,
I've answered you here:
http://stm32duino.com/viewtopic.php?f=29&t=2872&start=20#p37760

Linked to #150 .

fpistm, 2017-11-29:
I've tested with this sketch and by adding WEAK to the `void TIM6_IRQHandler(void)` here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/73dd420043b1d83a62d64877e7e4845971d6e366/cores/arduino/stm32/timer.c#L1119

```
void tim6_init()
{
RCC->APB1ENR |= RCC_APB1ENR_TIM6EN; // Enable TIM6 clock
TIM6->PSC = 41999; // Set prescaler to 41999
TIM6->ARR = 5999; // Set auto-reload to 5999
TIM6->CR1 |= TIM_CR1_OPM; // One pulse mode
TIM6->EGR |= TIM_EGR_UG; // Force update
TIM6->SR &= ~TIM_SR_UIF; // Clear the update flag
TIM6->DIER |= TIM_DIER_UIE; // Enable interrupt on update event
NVIC_EnableIRQ(TIM6_DAC_IRQn); // Enable TIM6 IRQ
TIM6->CR1 |= TIM_CR1_CEN; // Enable TIM6 counter
}
extern "C" void TIM6_DAC_IRQHandler()
{
if (TIM6->SR & TIM_SR_UIF != 0) // If update flag is set
{
TIM6->SR &= ~TIM_SR_UIF; // Interrupt has been handled
Serial.println("INTERRUPT");
}
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  tim6_init();
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

fpistm, 2017-11-29:
Closed as issue understood by submitter (See forum).
Moreover, it will be handle during Timer review.

closed 2017-11-29 by fpistm