stm32duino/Arduino_Core_STM32 #163: vsnprintf_P was not declared

DuyTrandeLion, 2017-11-30:
Today I used a library and suddenly I received an error 
error: 'vsnprintf_P' was not declared in this scope

  vsnprintf_P (cmdBuf, CMD_BUFFER_SIZE, (char*)cmd, args);

So can this function be added in the core? Thank you

fpistm, 2017-11-30:
Hi,
Which library?
Those functions are part of the AVR lib.
http://www.atmel.com/webdoc/avrlibcreferencemanual/group__avr__stdio.html

All those *_P functions are intend to work  with flash instead SRAM.
Currently, don't know if those functions should be added to this core ar at least reference them and point to standard one.

Ane suggestion/feedback are welcom

DuyTrandeLion, 2017-11-30:
Hi, thank you for your reply.

I use the WiFiEsp library https://github.com/bportaluri/WiFiEsp
This library helps controlling the esp8266 with AT command easier. 

I think making a similar function is still oke then. But I don't know how the function works.

fprwi6labs, 2017-11-30:
Hi,
I use a tiny vsnprintf implementation from ST cube in the new LoRa library (see this [PR](https://github.com/stm32duino/I-NUCLEO-LRWAN1/pull/1))
@DuyTrandeLion I let you to look at the code and try.

DuyTrandeLion, 2017-11-30:
Thankss I will report the result later. :D

DuyTrandeLion, 2017-11-30:
The sketch compiled successfully but I haven't tested all function. Oh I am using the Nucleo-L432KC, so can I use Serial1? I saw that I can enable USART1 to 3. 
And about ADC, does the ADC automatically generate an interrupt when I call the function? I wonder if it works like using the STM32CubeMX

fpistm, 2017-11-30:
Yes you can use the Serial1. By selecting the '_Enable first third_' in '_Serial interface'_ menu, Serial1 will be instantiate. 
By default, the Rx/Tx used will be the first one found in the PinMap_UART_RX[]/TX[] in [PeripheralPins.c](https://github.com/stm32duino/Arduino_Core_STM32/blob/d09167fbb5a5d36992d90a8135d07108fdf3e93f/variants/NUCLEO_L432KC/PeripheralPins.c#L117) file for USART1 peripheral.
So, for L432: PA10/PA9.
You could change it before call `Serial1.begin()` thanks:
```
Serial1.setRx(PB7);
Serial1.setTx(PB6);
```
About ADC, If you talk about `analogRead()`, there is no interrupt.
We are currently working on extended ADC API.

PR #166 (2017-12-05): Updated pgmspace.h
Based on Paul Stoffregen work on Teensy 3 (http://pjrc.com)

Fix #163
commit b346f699 (2017-12-05): Updated pgmspace.h

Based on Paul Stoffregen work on Teensy 3 (http://pjrc.com)

Fix #163
closed 2017-12-05 by fpistm