stm32duino/Arduino_Core_STM32 #165: New Board: STM32F030F4 Demo Board

Adminius, 2017-12-04:
Hi,
i'm trying to add support for this china board.
[image]

1. Question: I think nobody will have "custom" boards in official Core. Right? => how to share this board with others?
2. STM32CubeMX doen't export stm32f030x4.h... so i use stm32f030x6.h instead, is it ok?
3. I tried to compile Blink.ino, but: `Sketch uses 10124 bytes (61%) of program storage space. Maximum is 16384 bytes.
Global variables use 3704 bytes (90%) of dynamic memory, leaving 392 bytes for local variables. Maximum is 4096 bytes.
Low memory available, stability problems may occur.`
Any chance to get more RAM?

fpistm, 2017-12-04:
Hi,
1/ it is possible to add custom boards. This core mainly manage MCU. Boards is only a dedicated "pinmapping".  (ex #61 I have to end).
2/ I think you should use this one: STM32F030F4Px.xml
(if this is this one http://www.hotmcu.com/stm32f030f4p6-minimum-systerm-boardcortexm0-p-208.html)
3. You could try to disable some instance (ex: No Serial)
But doing more stuff than a blink will not use a lot of more RAM as instance will be already here.

fpistm, 2018-01-09:
Hi @Adminius,
is it ok to close this issue?

Adminius, 2018-01-09:
@fpistm pls not yet. i was on vacation. 
it's ready, but i have to test it.

fpistm, 2018-01-09:
No worries ;)

Adminius, 2018-01-18:
btw. this board/issue is not blocker for next release ;)

Sod-Almighty, 2018-03-04:
I have this board. How can I use it with this library? I don't see any options to select this model of chip.

Adminius, 2018-03-05:
it is not possible yet... i have to find a time to test my implementation. after that i can commit it to this repository.

P.S. what are you thinking about this PIN-Order?
[image]

This board have not enough pins for standart Arduino Layout (e.g. SPI on D10-D13)
I also "converted" UART-Header to i2c header, because i2c is only available on this pins, but UART can be switched to PA2/PA3

fpistm, 2018-03-05:
I think you should not try to be aligned with Ardunio layout for this board, user will mainly use the pin name PXy  (like the blue pill)

Adminius, 2018-03-05:
i will also use PXy ;)
I thing i will change the order... 
PA0 - D0 - A0
PA1 - D1 - A1
PA2 - D2 - A2 - TX
PA3 - D3 - A3 - RX
PA4 - D4 - A4 - LED
PA5 - D5 - A5 - SCK
PA6 - D6 - A6 - MISO
PA7 - D7 - A7 - MOSI
PB1 - D8 - A8 - SCK
PA9 - D9 - SCL
PA10 - D10 - SDA
PA13 - D11 - SWDIO
PA14 - D12 - SWCLK
it makes more sense an Dy and Ay are consist (also with PXy)

fpistm, 2018-03-05:
Just pay attention you will have to duplicate Analog pins else this will not work.
See https://github.com/stm32duino/Arduino_Core_STM32/commit/ffe5a3ff851c2e4e33348848fa4c976bc6267a55

I do that for Maple variant:
https://github.com/stm32duino/Arduino_Core_STM32/pull/213/files#diff-a15dd2dfb958a07755b98c8a5767f968R76

Adminius, 2018-03-05:
yes, i know:

> // Pin number
enum {
  PA0,  //D0 
  PA1,  //D1 
  PA2,  //D2  - RX
  PA3,  //D3  - TX
  PA4,  //D4  - LED
  PA5,  //D5  - SCK
  PA6,  //D6  - MISO
  PA7,  //D7  - MOSI
  PB1,  //D8  - SS
  PA9,  //D9  - SCL (TX UART Header)
  PA10, //D10 - SDA (RX UART Header)
  PA13, //D11 - SWDIO
  PA14, //D12 - SWCLK
 // Duplicated pins in order to be aligned with PinMap_ADC
 // A0 have to be greater than NUM_ANALOG_INPUTS 
  PA_0,  //D13/A0 = D0
  PA_1,  //D14/A1 = D1
  PA_2,  //D15/A2 = D2
  PA_3,  //D16/A3 = D3
  PA_4,  //D17/A4 = D4
  PA_5,  //D18/A5 = D5
  PA_6,  //D19/A6 = D6
  PA_7,  //D20/A7 = D7
  PB_1,  //D21/A8 = D8
};

fpistm, 2018-03-06:
That's a good question.
The silkscreen is:
> STM32F030F4 Demo board

So maybe it could be the name in the menu.

For variant name, something like that:
> DEMO_F030F4
> GENERIC_F030F4
> MSB_F030F4 (for Minimum System Board)

For the menu, maybe add a menu "Other"

romainreignier, 2018-03-06:
Don't forget that there is another variant of this board with a **mini USB** connector and a different pinout.

[image]

And also this smaller version that look like a cut BluePill

[image]

So, it will be a bit tricky to name those boards.

fpistm, 2018-03-06:
So for the menu: 
_Generic F030 series_

Adminius, 2018-03-06:
i would say, "STM32F103 series" should be "F103 series":
[image]

fpistm, 2018-03-06:
No you do not need pid/vid if no usb

Adminius, 2018-03-07:
i need help :( 
i can compile, but i get a lot of warnings like:

> C:\Users\Eugen\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2018.1.18\system/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_ll_tim.h: In function 'void LL_TIM_EnableBRK(TIM_TypeDef*)':

> C:\Users\Eugen\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2018.1.18\system/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_ll_tim.h:2790:10: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'

> (void)(tmpreg);

my configuration:
[STM32F030F4P6.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/1788564/STM32F030F4P6.zip)

i don't know what is wrong...

P.S. HW Upload not tested yet...

fpistm, 2018-03-07:
No worries, it is only warnings in LL.
I do not fix those warnings as I do not change HAL/LL code. 
This will be fixed with next(s) F0 HAL update from the Cube.
I already ignore several warning.
https://github.com/stm32duino/Arduino_Core_STM32/blob/b113598ed8bc6cae889144363fcf9c43c5c0f02d/cores/arduino/stm32/stm32_def.h#L55

I review your variant. Nice work ;)
One remark about the board.txt
You can add the GenF030.menu.xserial:
```
GenF030.menu.xserial.generic=Generic Serial
GenF030.menu.xserial.none=No Serial
GenF030.menu.xserial.none.build.xSerial=-DNO_HWSERIAL
GenF030.menu.xserial.all=All
GenF030.menu.xserial.all.build.xSerial=-DALL_HWSERIAL
```
Only 3 options as only 3 U(S)ART could be available for F0

Adminius, 2018-03-07:
oh, thanks. i didn't rework this section because of warnings...
F030F4 hat only one USART1...

fpistm, 2018-03-07:
I know for F030F4 but F030x8/C have 2 USART ;)
So as it is for GenericF030 ;)

Adminius, 2018-03-15:
Upload with ST-Link:
Blink LED - OK!
PWM LED -> region `FLASH' overflowed by 1360 bytes

fpistm, 2018-03-16:
I've use your previous code to test and reproduce the same. I'm not surprise.
There is several way to decrease overhead.
First one is to use LL to do the SystemClock_Config() because 
HAL_RCC_OscConfig uses around 964 bytes !

I've generate the code with CubeMx to use LL instead of HAL for RCC:
```
WEAK void SystemClock_Config(void)
{

  LL_FLASH_SetLatency(LL_FLASH_LATENCY_1);

  if(LL_FLASH_GetLatency() != LL_FLASH_LATENCY_1) {
    Error_Handler();
  }
  LL_RCC_HSI_Enable();

   /* Wait till HSI is ready */
  while(LL_RCC_HSI_IsReady() != 1) {
  }
  LL_RCC_HSI_SetCalibTrimming(16);

  LL_RCC_HSI14_Enable();

   /* Wait till HSI14 is ready */
  while(LL_RCC_HSI14_IsReady() != 1) {
  }
  LL_RCC_HSI14_SetCalibTrimming(16);

  LL_RCC_PLL_ConfigDomain_SYS(LL_RCC_PLLSOURCE_HSI_DIV_2, LL_RCC_PLL_MUL_12);

  LL_RCC_PLL_Enable();

   /* Wait till PLL is ready */
  while(LL_RCC_PLL_IsReady() != 1) {
  }
  LL_RCC_SetAHBPrescaler(LL_RCC_SYSCLK_DIV_1);

  LL_RCC_SetAPB1Prescaler(LL_RCC_APB1_DIV_1);

  LL_RCC_SetSysClkSource(LL_RCC_SYS_CLKSOURCE_PLL);

   /* Wait till System clock is ready */
  while(LL_RCC_GetSysClkSource() != LL_RCC_SYS_CLKSOURCE_STATUS_PLL) {
  }
  LL_Init1msTick(48000000);

  LL_SYSTICK_SetClkSource(LL_SYSTICK_CLKSOURCE_HCLK);

  LL_SetSystemCoreClock(48000000);

  LL_RCC_HSI14_EnableADCControl();

  LL_RCC_SetUSARTClockSource(LL_RCC_USART1_CLKSOURCE_PCLK1);

  LL_RCC_SetI2CClockSource(LL_RCC_I2C1_CLKSOURCE_SYSCLK);

  /* SysTick_IRQn interrupt configuration */
  NVIC_SetPriority(SysTick_IRQn, 0);
}

```
Now building Fading sketch with default optim (-Os) I've had `region `FLASH' overflowed by 64 bytes`
Note that I've configure to have SYSCLK at 48MHz. Your config set it to 8MHz.
After that I've disabled the Serial as not use in this sketch, then:
Sketch uses 15632 bytes (95%) of program storage space. Maximum is 16384 bytes.

After that, I think it should be best to disable some unused feature when not required.
Example here  I2C, SPI by disabling the HAL but currently I do not add the code using it under switch.
Example for I2C in twi.h if HAL_I2C_MODULE_ENABLED is not defined then disabled the code.
I've made a quick try by adding the switch:
```
Sketch uses 13072 bytes (79%) of program storage space. Maximum is 16384 bytes.
Global variables use 996 bytes (24%) of dynamic memory, leaving 3100 bytes for local variables. Maximum is 4096 bytes.
```
I think, will  open an issue to be able to disabled some of basic feature when not used in the sketch.

Adminius, 2018-03-16:
thanks!
CubeMX says SYSCLK 48Mhz:
[image]

fpistm, 2018-03-16:
For your current config ?

Adminius, 2018-03-16:
for any internal function (SPI via DMA or so).
i don't know what CRC is good for...
i don'T no if USART depends UART_MODULE...

fpistm, 2018-03-16:
You could comment HAL_USART_MODULE_ENABLED
Serial use only asynchronous.

/* CRC FEATURE: Use to activate CRC feature inside HAL SPI Driver
* Activated: CRC code is present inside driver
* Deactivated: CRC code cleaned from driver
*/

#define USE_SPI_CRC                     0U
So I think HAL_CRC_MODULE_ENABLED could be disabled.
Anyway as CRC and USART function are not used, code is already removed.

huangguozhen, 2018-06-07:
@Adminius do you solve it.
program `stm32f030f4p6` with arduino.

Adminius, 2018-06-07:
@huangguozhen no :( 
i'm waiting for #228 

Actually, after windows reinstall, i have troubles with ST-Link (with all variants, arduino says: successfully uploaded, but there nothing happens)... so i have to check it before i can work with F030

huangguozhen, 2018-06-07:
@Adminius I am also looking forward to it when i have a `stm32f030f4p6 demo board`.

huangguozhen, 2018-06-13:
@Adminius can you send me a `stm32f030f4p6 demo` config which work well.

Adminius, 2018-06-13:
https://github.com/stm32duino/Arduino_Core_STM32/files/1788564/STM32F030F4P6.zip

Adminius, 2018-08-12:
Hi @fpistm sorry for delay.
now i have time for this board again.
First of all, i cannot start this board with LL driver. Blink doesn't blink.
I've tested with my config:
```cpp
void SystemClock_Config(void)
{

  LL_FLASH_SetLatency(LL_FLASH_LATENCY_1);

  if(LL_FLASH_GetLatency() != LL_FLASH_LATENCY_1)
  {
  Error_Handler();  
  }
  LL_RCC_HSE_Enable();

   /* Wait till HSE is ready */
  while(LL_RCC_HSE_IsReady() != 1)
  {
    
  }
  LL_RCC_HSI14_Enable();

   /* Wait till HSI14 is ready */
  while(LL_RCC_HSI14_IsReady() != 1)
  {
    
  }
  LL_RCC_HSI14_SetCalibTrimming(16);

  LL_RCC_PLL_ConfigDomain_SYS(LL_RCC_PLLSOURCE_HSE_DIV_1, LL_RCC_PLL_MUL_6);

  LL_RCC_PLL_Enable();

   /* Wait till PLL is ready */
  while(LL_RCC_PLL_IsReady() != 1)
  {
    
  }
  LL_RCC_SetAHBPrescaler(LL_RCC_SYSCLK_DIV_1);

  LL_RCC_SetAPB1Prescaler(LL_RCC_APB1_DIV_1);

  LL_RCC_SetSysClkSource(LL_RCC_SYS_CLKSOURCE_PLL);

   /* Wait till System clock is ready */
  while(LL_RCC_GetSysClkSource() != LL_RCC_SYS_CLKSOURCE_STATUS_PLL)
  {
  
  }
  LL_Init1msTick(48000000);

  LL_SYSTICK_SetClkSource(LL_SYSTICK_CLKSOURCE_HCLK);

  LL_SetSystemCoreClock(48000000);

  LL_RCC_HSI14_EnableADCControl();

  LL_RCC_SetUSARTClockSource(LL_RCC_USART1_CLKSOURCE_PCLK1);

  LL_RCC_SetI2CClockSource(LL_RCC_I2C1_CLKSOURCE_SYSCLK);

  /* SysTick_IRQn interrupt configuration */
  NVIC_SetPriority(SysTick_IRQn, 0);
}
```
and [your's](https://github.com/stm32duino/Arduino_Core_STM32/issues/165#issuecomment-373646769)

both dosn't work. (It compiles without any problems but doesn't blink)

Your config doesn't use external 8MHz Crystal (HSE), but HSI internal... 

Should i add or change something to be able to use LL?

If i try to compile Fade-Example:
```
C:\Users\Eugen\AppData\Local\Temp\arduino_cache_891323\core\core_STM32_stm32_GenF030_pnum_DEMO_F030F4,upload_method_STLinkMethod,xserial_disabled,opt_osstd_0a5fea942f33be06b8c57e4f454faeda.a(syscalls_stm32.c.o): In function `_write':
syscalls_stm32.c:(.text._write+0x6): undefined reference to `uart_debug_write'
collect2.exe: error: ld returned 1 exit status
```
I think we have to give up... this MCU has to many restrictions :(

fpistm, 2018-08-13:
Since your update I've made some change in Serial to be able to disable it.
I think you're not aligned with the master.
Your issue is linked to this, mainly:
https://github.com/stm32duino/Arduino_Core_STM32/commit/98efbe71d25dcb8bb7d998619e88834485c60341

About LL,  I do not use it if they are not exists. On Nucleo there is often the HSE provided through MCO from ST-LINK but I preferred use HSE when a real HSE mounted on the board.

I guess for your issue it miss a config (power a block or something like that).

I can test with the disco F030. For HSE,

fpistm, 2018-08-13:
The fix for this issue is not in the 1.3.0.
You have it only when a printf is called in the code

aryeguetta, 2018-10-03:
Hi, I have read this thread and did my best to try and follow everything... 
I would be VERY happy to know if there is any progress with the Support for this board with Arduino IDE?
I have already had this board at hand and I would be happy to try Arduino IDE to develop on it.

I am interested in making this board supported on the Arduino IDE and I am sure that Professional like yourselves can DO that and I will be help :)

Please let me know if this board already been supported with Arduino IDE and if not why? and how Can I assist? thanks

Kaitoukid93, 2018-10-16:
I added this board to arduino but when I try to compile, an error show up. The arduino.h was used instead of stm32
how can I fix this. thanks
`\AppData\Local\Temp\arduino_build_687106\sketch\ASCIITable.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

 #include <Arduino.h>

                     ^

compilation terminated.

exit status 1
Error compiling for board Generic STM32F030 series.`

BLavery, 2018-10-29:
I have played with the customised board profile for STM32F4P6. I had been following similar investigation on my own, and then I discovered this thread.  My version was almost identical, exc I set SERIAL_UART_INSTANCE as 0, and it then save some compile bytes.

**In general, this board compiles under Arduino IDE, and it transfers using ST-link-V2 cheapie from eBay.**

Clearly it's short of flash. It copes happily with simple blink, anything using math, digital GPIO, analog read (I think - no error at least), shiftOut, delay, millis. 
It crashes with flash memory shortage on more substantial calls: Serial (when I still had that included), analogWrite, I2S, SPI, Servo.
So no terminal, no display of any sort, no PWM for motors.

**Possibilities:**

- ShiftOut as a SPI substitute for a simplified LCD module.
- Bitbang Serial out, even no buffering, for a very primitive terminal output.  Without print/printf. Ouch.
- Bitbanged PWM.  It is a moderate speed CPU.
- But there are other boards without this flash shortage. For this one, stick to digital IO and analog read & math, and enjoy its tiny price.

fpistm, 2018-11-02:
This board have been added thanks #367 .
@BLavery
there is several code size reduction with the latest release 1.4.0. 
Morever you can use the LTO optimization thanks the menu and also fully disabling the Serial if not requested.
There will be also more action to continue to reduce size, See #228.
I close this issue has the original request has been achieve thanks @Adminius

closed 2018-11-02 by fpistm

BLavery, 2018-11-07:
I have put up some library code for F103F4P6 here:
https://github.com/BLavery/STM32F030F4P6-Arduino
There are bit-banged versions of Serial and Wire that can compile into small code, to allow serial terminal and I2C OLED display for this board. Rough but working.

BLavery, 2018-11-08:
Aryeguetta,  As I understand it, those are USB identification "vendor" / "product" codes. This chip / board has no USB. The USB connector is only for power.

aryeguetta, 2018-11-09:
[image]
Hi, I have just wanted to be sure that the image I have assembeld is OK and I can put as reference for thast board.
I would be happy to find the Version 2 of this board mentioned above, a different pinout and also the small one that was refered in here as Bluepill 030 - I would be happy to buy and test them as well.
I am so happy that we have that board working under Arduino IDE - I thank you all for it.

fpistm, 2018-11-09:
The only issue I saw about pin numbering is the one raised by @BLavery about PA13 and PA14.
Which should be D11 and D12.
You could add SPI :
```
#define PA5  5  //D5/A5 - SCK
#define PA6  6  //D6/A6 - MISO
#define PA7  7  //D7/A7 - MOSI
#define PB1  8  //D8/A8 - S
```
And I2C on:
```
#define PA9  9  //D9    - SCL (TX UART header)
#define PA10 10 //D10   - SDA (RX UART header)
```
And by default Serial instance are configured on:

Adminius, 2018-11-09:
@BLavery 
D11-D12 on PFx doen't make sence, because you can't use this pins:
https://github.com/BLavery/STM32F030F4P6-Arduino/blob/master/(stm32)/1.4.0/variants/DEMO_F030F4/variant.cpp#L52-L53
That's why i used this mapping: PA13=D11 and PA14=D12
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/DEMO_F030F4/variant.cpp#L55-L56

I've made SDA and SCL on UART header, because SDA and SCL are available on this pins only...
RX/TX can be on PA9/PA10 or PA2/PA3. So that's why i mappend UART to PA2/PA3.

@aryeguetta you image is not applicable for official core (v1.5.0+),

BLavery, 2018-11-09:
Thanks @Adminius 
Your project, and you must make the correct long-term tech choices as you judge correct. All is good.

aryeguetta, 2018-11-11:
Thanks for letting me know about mismatches with the Official Core V1.5.
I have made the changes and hopefully, it will now be a GOOD reference for that board.
[image]
Now it might be applicable to Version 1.5.0+ :)
Thanks for Everything!!
Arye

fpistm, 2018-11-11:
@aryeguetta in fact it is ok for this core but as you mark it is for @BLavery  variant this is not correct as his variant define PF0 and PF1. So in fact, there is 2 versions of the picture: one for @BLavery  and one for 1.5.
 https://github.com/BLavery/STM32F030F4P6-Arduino/blob/9cee86c7261845b1f47615a0ff467fd215b5a71f/(stm32)/1.4.0/variants/DEMO_F030F4/variant.h#L59

aryeguetta, 2018-11-11:
Thanks fpistm, Point taken - made the appropriate changes and now I have two version for reference.
[image]
Hope it is Ok now :) thanks.

I do have another board related question - I have seen a version 2 of the Board itself for sell on AliExpress and I wonder:

1. Is this board differs a lot from version 2? if so, How? Besides the obvious micro USB vs mini USB power drive, Mirrored and flipped :)

1. Is it worth buying? Only for learning the schematics of the board and understandning the diferences between two version - if exists? There are three jumpers on the board, I assume it will add more functionality.

1. The have **not soldered an oscillator** to the board - meaning I can solder a different one if I want? Or I need to suply it with the one they mentioned in the image? Assuming it is 8MHz like "version 1".
**BTW:** I am not 100% sure they provide the 8MHz oscillator - it looks like we need to buy it ourselves, or isn't it? I hope that they do provide :) 
Here is the board I am talking about: (It is described as)

> ARM M0 STM32F030F4P6 development board / core learning board small system can be TTL serial download
[image]
p.s. Is ver2 a true 5V? How can we provide it with a 5V? from the ST-Link V2 or FTDI connections I see only 3v3. That means we MUST have the mini USB connected to provide the 5V otherwise we will not be able to use it, correct?
Thanks,
Arye

romainreignier, 2018-11-11:
I own all of the three boards mentioned above (but did not really used them
apart from a blink to test they work).

So the mini USB version as a 8 MHz quartz and from what I remember, the LED
is not connected to the same pin.

<https://user-images.githubusercontent.com/20950920/48310324-4927cb80-e596-11e8-9a11-e4650a2e0309.jpg>

   2.

   3.

<https://user-images.githubusercontent.com/20950920/48310410-6b6e1900-e597-11e8-9213-e318a999511b.jpg>

aryeguetta, 2018-11-11:
> I own all of the three boards mentioned above (but did not really used them apart from a blink to test they work). So the mini USB version as an 8 MHz quartz and from what I remember, the LED is not connected to the same pin.

Thanks for letting me know.
I would be happy to know where you get the other two - links would be great.
Did you get the Crystal with the version 2 board as well?
I am planning to get the 2nd version and I would be happy to know where I should get it and if it comes with the Crystal - I might need to solder it but it is OK that will cheapen up the price of the board as well.

The third one I have NOT even found on AliExpress and I am sure it would teach me the same way version 2 will or even more.
I would be happy to know more about those two other boards and also where I should be able to get them (cheap if possible) - I am ordering today since it is 11-11 and there are sales on AliExpress.
Please let me know.
Thank You!

fpistm, 2018-11-11:
@aryeguetta I do not see any difference btw the 2 images?
About other version of the board I do not know. I have not this board. It seems they are several distributors with small differences.

romainreignier, 2018-11-11:
So, yes I got the crystal with the board. I had to solder it.

About the links, I have the smallest one but it is now unavailable (I bought it the 26th of August 2016): https://www.aliexpress.com/item/STM32F030F4P6-minimum-system-board-core-board-learning-board-ARM-STM32/32585222766.html

But for the mini USB one, bought the same day, the page does not exist anymore: https://www.aliexpress.com/item/STM32F030F4P6-development-board-TTL-serial-ports-to-download/32353640682.html

All the 3 boards:

[image]

So if is not possible to find them anymore, maybe their support can be skipped.

romainreignier, 2018-11-11:
Anyway, I have found the mini USB one on Aliexpress: https://www.aliexpress.com/item/STM32F030F4P6-Board-3-3V-5V-Power-STM32-Minimum-System-Development-Board/32901427670.html

romainreignier, 2018-11-11:
And the small one but more expensive:

- https://www.aliexpress.com/item/STM32F030F4P6-minimum-system-board-core-board-learning-board-ARM-STM32/1856470983.html

[image]

And other variants but much more expensive:

- https://www.aliexpress.com/item/STM32F030F4P6-development-board-M0-development-board-USB-to-serial-port-support-ISP-download/32884815397.html

[image]

- https://www.aliexpress.com/item/STM32F030F4P6-core-development-board-small-system-board-M0-kernel/32880134049.html

[image]

- https://www.aliexpress.com/item/Fast-Free-Ship-ARM-M0-STM32F030F4P6-development-board-core-learning-board-minimum-system-can-be-TTL/32808685403.html

[image]

aryeguetta, 2018-11-11:
Thanks, Romain, The ONLY one that I will buy is the one w/out the Crystal - I will solder it myself as well.
All other boards ARE way too expensive to be used with ARM Cortex M0 - that is NOT the way I see to invest. I want to be able to put a board that will replace Arduino UNO R3 or the BluePill with projects that do NOT need a lot of coding and use mathematics and digital/Analog basics. 
That way I will not waste a bigger board in small projects and the prices for that Demo board are not expensive.
I have already ordered a couple of Demo boards - "version one" less then 1.50 USD each and that is a good price. Now that I CAN program them with Arduino IDE - It is better!!!!
I will get the version 2 and hopefully will look for a cheap version of the "Short" Blue Pill look alike board. I hope less than 2 USD as well. 

I would be happy to learn more about these boards and build with them even to eliminate the use of ATtiny85 - where needed.
Thanks for all the Links - all other boards are WAY expensive 15 USD for CORTEX M0 - I don't think I would spend that much of money for it. And frankly we are NOT getting much for that price.

Thanks fpistm,
Regarding the two versions:
**Yes You may use BOTH of them as you like.** and i would be happy if you do so.
> - **STM32 Core version:**

[image]

> - **Blavery Version is:**

[image]

That will leave us with two different images - of which i am not sure there will be a need for both eventually. 
I will get the version 2 and after i will get it I will work on the board and try making it to work with Arduino IDE. I wish i will be able to get the small one cheap.

[image]

If someone knows where I could find a CHEAP version of this Short STM32 I would be happy to know.

Thanks,
Arye

kgamecarter, 2019-06-17:
some cheap version STM32 
[https://item.taobao.com/item.htm?id=36481303532](https://item.taobao.com/item.htm?id=36481303532)
[https://item.taobao.com/item.htm?id=590471216394](https://item.taobao.com/item.htm?id=590471216394)