stm32duino/Arduino_Core_STM32 #179: I2C on the F411RE

frenchie68, 2017-12-15:
Hi dear STM32 afficionados,

I have been trying to get I2C to work on a Nucleo64 STM32F411RE for about two
weeks now. I don't have an oscilloscope but I read the datasheet for the board
and it specifies a 4.7K pullup resistor to Vdd (3.3V) on SCLK and SDA--check.
On the F441xE board, this maps to pins 15 (PB8) and 14 (PB9) respectively.
The datasheet also mentions a current limiting resistor of an unspecified
value--I went for 270 Ohms.

Well, for starters, those pins are not replicated on a standard Arduino shield,
you'd have to access them on the ST-Morpho connector but that's not my main
concern. My issue is with the pre-encoded I2C speeds. On page 106 the datasheet
states:

SCL  (kHz)	I2C_CCR value
400			0x8019
300			0x8021
200			0x8032
100			0x0096
50			0x012C
20			0x02EE

However, in https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/twi.h we have the following:

#else //STM32F4xx
  I2C_10KHz =   10000,
  I2C_50KHz =   50000,
  I2C_100KHz =  100000,
  I2C_200KHz =  200000,
  I2C_400KHz =  400000,
/*  I2C_600KHz =  600000,
  I2C_800KHz =  800000,
  I2C_1000KHz = 1000000*/ //Not supported
#endif

How come?

        Francois Laagel

LMESTM, 2017-12-18:
HI @frenchie68 
What is exactly your concern ? Do you need a specific frequency that is not coded in the twi.h file ? 
I agree that the examples frequencies and registers values that are listed in datasheets are not exactly the ones that are coded, but is this really a problem? Developper has selected a list of frequencies that seemed useful. Typically here, it was decided that 300kHz is not a frequency that is often used so it was not listed, but if you think this is a problem, please let us know.

fprwi6labs, 2017-12-18:
Hi @frenchie68 

For F4 boards the I2C_CCR is calculated from the I2C speed and the duty cycle.
I invite you to look at stm32f4xx_hal_i2c.c at line 405.

I don't think it is a cause of your issue.

frenchie68, 2017-12-18:
Hi guys,

OK, the F4 is treated differently than other boards and I had missed that. I'll have to have a closer look at the code. My problem must be electrical in nature. Sorry about this.

closed 2017-12-18 by fpistm

frenchie68, 2017-12-28:
Gentlemen,

This issue might be closed but I think I might be able to contribute a small and useful change that would make the standard I2C scanner code more realistic. In twi.c, the bus frequency is conservatively set to 100KHz in i2c_custom_init but GPIO_InitStruct.Speed is set to GPIO_SPEED_FREQ_HIGH. The reference manual for the STM32F411RE links me back to the datasheet for that quantity, and there is no such data there. However, experience shows that a setting of GPIO_SPEED_FREQ_LOW does not produce "ghost" devices, whereas the current setting does.

https://github.com/stm32duino/Arduino_Core_STM32/blob/e144d837a9c7b705a57f90535fd2794d3574ab99/system/Drivers/STM32L1xx_HAL_Driver/Inc/stm32l1xx_hal_gpio.h#L161 would indicate that I am on the right path.

You might want to look into it.

        Francois