stm32duino/Arduino_Core_STM32 #180: New variant STM32F303CCT6 / USB VCP

Hoel, 2017-12-17:
Hello,
I am making new variants definitions for boards we make and i would like to know how to get the USB serial (USB VCP) working for them, at first STM32F303CC, once done STM32F103L151C8, STM32F103CB, STM32F411CE, STM32L476RG. 
Apparently some rework has been ongoing on USB VCP for STM32F103CB (bluepill) few month ago and bluepill will be part of the next release, is it posible to get bluepill variant files so i can start from this for USB VCP on STM32F303CC?
Thanks

fpistm, 2017-12-18:
Hi, 
since this PR core has got lot of change. So, I doubt it is compatible. I have to rebase it properly.

Hoel, 2017-12-18:
Hi, thanks for the confirmation. Sadly serial over USB doesnt work either on the bluepill F103 and maple mini variants from your repo. 
This fork https://github.com/danieleff/STM32GENERIC/issues/52 has the USB working for F103, i cannot get it to work for F303 either.

Hoel, 2017-12-22:
The problem is now solved, i used a fork of this repo https://github.com/huaweiwx/STM32GENERIC  USB virtual serial works on the F303

closed 2017-12-22 by Hoel