stm32duino/Arduino_Core_STM32 #182: Yes

JuanCarranza-Borra, 2017-12-19:
Please, before reporting any issue:
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported or discussed on the [stm32duino forum](http://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the Arduino_Core_STM32.
- **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/sugestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

When reporting any issue, please try to provide all relevant information:

- Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)
- Which STM32 based board you used
- If other hardware is required to reproduce the issue, please provide links or part numbers where this hardware can be purchased.
- Which version of STM32 core and Arduino, Tools menu settings, operating system, ...

Thanks

fpistm, 2017-12-19:
Seems you made a wrong manipulation ;)

closed 2017-12-19 by fpistm

JuanCarranza-Borra, 2017-12-19:
I’m sorry, I don’t know how it hapenned ..