stm32duino/Arduino_Core_STM32 #184: DFU (maple-upload) bootloader

Hoel, 2017-12-22:
I saw the post on http://www.stm32duino.com/viewtopic.php?f=3&t=2486&start=10
has any dfu bootloader been written (to use with malple-upload)? The bootloader here https://github.com/rogerclarkmelbourne/STM32duino-bootloader is based on direct regiter access and complicate makefile, i couldnt make a working SW4STM32 project from it (builds but binary doesnt work) and anyway it cannot be ported easily to STM32L1 nor STM32F3. A bootloader easy to tailor to the various STM32 targets would be a very good addition to the official STM32duino core.

fpistm, 2017-12-29:
Hi @Hoel ,
Maybe it could be a good addition but currently I have no bandwidth to investigate it.
Any contribution are welcome.

Hoel, 2017-12-29:
Hi,
OK fine, i made the changes in core to handle ROM bootloader for newer STM32 (such as F303CC) it works fine. Currently working on a 4K DFU 1.1 bootloader for other variants, it will be ready once the re-enumeration issue is fixed.

Testato, 2018-09-23:
+1 for this request.

Usb BL is most important on every board (arduino Uno is the most valuable example, without it Arduino will not arrive where it is arrived), and the Bluepill (F103) is the most widely sold among the ST maker boards.

so I hope this request scales the ST Arduino core priorities

fpistm, 2019-02-12:
This one is ongoing thanks @BennehBoy 
See #415

closed 2019-02-12 by fpistm