stm32duino/Arduino_Core_STM32 #186: Conflict Types ('unsigned char' and 'int')

Mirki, 2018-01-03:
During the compiling from I2Cdev\I2Cdev.cpp I'm getting the error message

E:\Dev\Arduino\libraries\I2Cdev\I2Cdev.cpp:276:75: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')

             for (uint8_t k = 0; k < length; k += min(length, BUFFER_LENGTH)) {

                                                                           ^
https://github.com/jrowberg/i2cdevlib

fpistm, 2018-01-03:
Hi @Mirki ,
The core use the STL C++.
min, max,... require to compare same type.
Seems related to this post:
http://stm32duino.com/viewtopic.php?f=2&t=3038&p=39369#p39369

The i2cdevlib needs some updates to avoid this kind of issue and harden the code.
`         for (uint8_t k = 0; k < length; k += min((uint32_t)length, BUFFER_LENGTH)) {`
You should open an issue to this repo to fix all min/max call.

Mirki, 2018-01-04:
sounds valid.
thanks

closed 2018-01-04 by Mirki

sgounane, 2018-01-25:
in i2cdev.c i replaced BUFFER_LENGTH whith "(unsigned char)32" and it works for redbearlab board

fpistm, 2018-01-25:
Yes, this one other possibility.
While the type of the two arguments is the same when using STL min/max, it's ok.
But be aware that BUFFER_LENGTH could be different than 32.
This is mainly the value used in current Arduino but it can be changed.