stm32duino/Arduino_Core_STM32 #189: Problem with external RTC on STM32L476RG board

mayukhIITD, 2018-01-03:
Dear All,

As you know STM32L476RG board has RTC but it is not connected to a Battery. Hence I would like to use an external RTC like ds1307. But when I use DS1307 (I2C address= 0x68) , the usual Arduino libraries like RTCLib  are giving output as:

2165/165/165 (Tuesday) 165:165:85
 since midnight 1/1/1970 = 2275406289s = 26335d
 now + 7d + 30s: 2042/2/15 5:48:15

The same code is running properly when I connect the DS1307 to Arduino board.
Any solution for this will be of much help...

Thanks,
Mayukh

LMESTM, 2018-01-05:
Hi @mayukhIITD 

Have you checked if I2C communication works ok ? 
If you are using the default example from here: https://github.com/adafruit/RTClib/blob/master/examples/ds1307/ds1307.ino
and can you share the full log ? 

I'd like to check if I2C runs fine with example code;
```
  if (! rtc.isrunning()) {
    Serial.println("RTC is NOT running!");
}
```
Also please make sure you are using the latest version of the core. 
Unfortunately I don't have a ds1307 with me to test ...

mayukhIITD, 2018-01-05:
Hi Laurent,

Thanks for your reply..
RTC is running . Otherwise it would have printed "RTC is NOT running!". RTC is running but showing wrong values. 

Also ,when I use I2Cscanner, the I2C port of DS1307 i.e 0x68 does not show up, whereas I2C addresses corresonding to HTS221, LSM303AGR,LSM303AGR,LSM6DSL,LPS22HB  which are on board in STM32L476RG board are showing up...

The following is the whole log when I run ds1307.ino.

Archiving built core (caching) in: /tmp/arduino_cache_334946/core/core_STM32_stm32_Nucleo_64_pnum_NUCLEO_L476RG,upload_method_STLink,xserial_third,usb_none,opt_osstd_d3552e88b625682bbc95021fcb00b8b6.a
Sketch uses 23064 bytes (2%) of program storage space. Maximum is 1048576 bytes.
Global variables use 3336 bytes (2%) of dynamic memory, leaving 127736 bytes for local variables. Maximum is 131072 bytes.
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: Loading device parameters....
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: Device connected is: L4 device, id 0x10076415
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: Attempting to write 23484 (0x5bbc) bytes to stm32 address: 134217728 (0x8000000)
st-flash 1.3.0
EraseFlash - Page:0x0 Size:0x800 EraseFlash - Page:0x1 Size:0x800 EraseFlash - Page:0x2 Size:0x800 
Flash page at addr: 0x08000000 erased
Flash page at addr: 0x08000800 erased
Flash page at addr: 0x08001000 erasedEraseFlash - Page:0x3 Size:0x800 EraseFlash - Page:0x4 Size:0x800 EraseFlash - Page:0x5 Size:0x800 EraseFlash - Page:0x6 Size:0x800 
Flash page at addr: 0x08001800 erased
Flash page at addr: 0x08002000 erased
Flash page at addr: 0x08002800 erased
Flash page at addr: 0x08003000 erasedEraseFlash - Page:0x7 Size:0x800 EraseFlash - Page:0x8 Size:0x800 EraseFlash - Page:0x9 Size:0x800 EraseFlash - Page:0xa Size:0x800 
Flash page at addr: 0x08003800 erased
Flash page at addr: 0x08004000 erased
Flash page at addr: 0x08004800 erased
Flash page at addr: 0x08005000 erasedEraseFlash - Page:0xb Size:0x800 2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: Finished erasing 12 pages of 2048 (0x800) bytes
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/common.c: Starting Flash write for F2/F4/L4
2018-01-05T14:51:38 INFO /local/data/work/stm32/repo/stlink/src/flash_loader.c: Successfully loaded flash loader in sram

Flash page at addr: 0x08005800 erased2018-01-05T14:51:39 INFO /local/data/work/stm32/repo/stlink/src/common.c: Starting verification of write complete
2018-01-05T14:51:39 INFO /local/data/work/stm32/repo/stlink/src/common.c: Flash written and verified! jolly good!

size: 23484

And on Serial Monitor it shows:

2165/165/165 (Tuesday) 165:165:85
since midnight 1/1/1970 = 2275406289s = 26335d
now + 7d + 30s: 2042/2/15 5:48:15

Please help...

Thanks,
Mayukh

fpistm, 2018-01-05:
Hi, I will have next week an external RTC based on the DS1307 then I could perform some test.

LMESTM, 2018-01-05:
@mayukhIITD 
Ok. Just to be sure, have you uncommented one of the below lines ?
https://github.com/adafruit/RTClib/blob/master/examples/ds1307/ds1307.ino#L21
or
https://github.com/adafruit/RTClib/blob/master/examples/ds1307/ds1307.ino#L23
?
As @fpistm stated, hopefully we can test next week.

mayukhIITD, 2018-01-06:
@LMESTM  Yes I have uncommented it. Adjusting time is also not working. Please let me know after you test it..

Thanks,
Mayukh

fpistm, 2018-01-08:
I've tested with this module Adafruit 3296:
https://www.adafruit.com/product/3296

**Setup**:
- Arduino-1.8.5
- Last official core release (2017.11.24) and current master
- Nucleo-L476RG
- Lateste RTCLib release (1.2.0) and master (https://github.com/adafruit/RTClib)
- Sketch: ds1307 (tested with both `rtc.adjust()`)

**Wiring**:
- GND 
- +5V
- SCL: D15
- SDA: D14
- SQW: NC

All is OK.
```
RTC is NOT running!
2018/1/8 (Monday) 11:46:0
 since midnight 1/1/1970 = 1515411960s = 17539d
 now + 7d + 30s: 2018/1/16 0:16:6

2018/1/8 (Monday) 11:46:2
 since midnight 1/1/1970 = 1515411962s = 17539d
 now + 7d + 30s: 2018/1/16 0:16:8

2018/1/8 (Monday) 11:46:5
 since midnight 1/1/1970 = 1515411965s = 17539d
 now + 7d + 30s: 2018/1/16 0:16:11

2018/1/8 (Monday) 11:46:8
 since midnight 1/1/1970 = 1515411968s = 17539d
 now + 7d + 30s: 2018/1/16 0:16:14

2018/1/8 (Monday) 11:46:11
 since midnight 1/1/1970 = 1515411971s = 17539d
 now + 7d + 30s: 2018/1/16 0:16:17
```

The value you've got is the default value when module is not reachable:
- wired issue
- without coin cell battery
- wrong I2C instance?

Which pins of the Nucleo L476 you used?

mayukhIITD, 2018-01-09:
Thank you so much Frederic @fpistm and Laurent  @LMESTM  for your help and clear explanation. The issue is resolved now!!
The problem was: I was using an Arduino Datalogger shield with RTC on it. And in that shield RTC SCL and SDA were connected to A5,A4 instead of D15,D14. That's why it was not working. I took a separate DS1307 and connected to D14,D15 of Nucleo L476 and the RTC is working fine..

Thank you so much. You guys are awesome!! :)

Regards,
Mayukh

fpistm, 2018-01-09:
Welcome
To use your other shield, you can change the variant.h file to add/uodate the specific SDA/SCL pins to use for Wire.
```
#define SDA A4
#define SCL A5
```

To add more flexibility, I will add new method to Wire library to be able to change default pin of Wire instance.
Something like that in Wire.h
```
 void setSDA(uint32_t sda) {_i2c.sda = digitalPinToPinName(sda);};
 void setSCL(uint32_t scl) {_i2c.scl = digitalPinToPinName(scl);};
```
Then at sketch level, before the call of `Wire.begin();`
Doing something like that:
```
Wire.setSDA(A4);
Wire.setSCL(A4);
Wire.begin(); // or  rtc.begin(); in this case which call the Wire.begin();
```

closed 2018-01-09 by fpistm

LMESTM, 2018-01-09:
@fpistm got the hardware and he did all the work :-)   Thanks to him !

fpistm, 2018-01-09:
New API added to give more flexibility #193. Long time ago I've got this change in my local repo.

lockie555, 2022-09-19:
Hi , 
My problem is that I get an "Error compiling for board Nucleo-64." using the Sketch: ds1307 code. I am using an STM32L152 MCU and an arduino RTC and SD card shield. My previous forays in using an 8563 RTC  with the STM32L152 also caused the same error but this was eventually solved by replacing "rtc" with "myrtc" as the Nucleo has its own RTC . This time, however, I tried doing the same but it made no difference. The only mention of error is in the RTC_DS1307.cpp source code (see below)
Any ideas??
Lockie555

pert. err code:
C:\Users\marts\Documents\Arduino\libraries\RTClib\src\RTC_DS1307.cpp: In member function 'void RTC_DS1307::writenvram(uint8_t, const uint8_t*, uint8_t)':
C:\Users\marts\Documents\Arduino\libraries\RTClib\src\RTC_DS1307.cpp:109:18: error: invalid conversion from 'const uint8_t*' {aka 'const unsigned char*'} to 'uint8_t*' {aka 'unsigned char*'} [-fpermissive]
  109 |   i2c_dev->write(buf, size, true, &addrByte, 1);
      |                  ^~~
      |                  |
      |                  const uint8_t* {aka const unsigned char*}
In file included from C:\Users\marts\Documents\Arduino\libraries\RTClib\src\RTClib.h:25,
                 from C:\Users\marts\Documents\Arduino\libraries\RTClib\src\RTC_DS1307.cpp:1:
C:\Users\marts\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_I2CDevice.h:15:23: note:   initializing argument 1 of 'bool Adafruit_I2CDevice::write(uint8_t*, size_t, bool, uint8_t*, size_t)'
   15 |   bool write(uint8_t *buffer, size_t len, bool stop = true,
      |              ~~~~~~~~~^~~~~~
Using library RTClib at version 2.1.1 in folder: C:\Users\marts\Documents\Arduino\libraries\RTClib 
Using library Adafruit_BusIO at version 1.3.3 in folder: C:\Users\marts\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0.0 in folder: C:\Users\marts\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\Wire 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\marts\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SrcWrapper 
Using library SPI at version 1.0.0 in folder: C:\Users\marts\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SPI 
exit status 1
Error compiling for board Nucleo-64.