stm32duino/Arduino_Core_STM32 #209: Upload via UART bootloader & stm32flash

fischermario, 2018-01-29:
I have this board:

[Picture](https://dzrmo.files.wordpress.com/2013/01/stm32f1s.png)

It has a STM32F103RBT6 on it. I want to program it using the UART bootloader via the RS232 serial port. I have tested this method by creating a simple blink program using Eclipse + GNU MCU Plugin + STCubeMX (HAL) + GNU ARM Toolchain. Uploading the resulting .bin file via the UART bootloader works flawlessly and the LED is blinking (as expected).

If I try the same thing (blinky) with the Arduino IDE I can upload the resulting .bin file, but it does not do anything. I have selected Nucleo F103RB because it seems like the closest configuration for my board. I have also changed the LED pin to D15 because my LED is connected to PB8. I have also tried to output some data on the serial port, but did not get any results.

I have also tried to collect all the necessary files from the STM32 core and put it into an Eclipse project. It builds successfully (after some modifications), but the result is the same.

After a lot of hair-pulling I am giving up and I am now asking here. What am I missing?

PS.: As advised I would rather have posted this on the stm32duino.com forum, but I seem to have no rights to create a new thread anywhere (though I am a registered user).

fpistm, 2018-01-29:
Hi @fischermario 
probably an issue with the clock config.
The Nucleo F103RB is configured to use MCO from ST-LINK (as HSE).
So if your board does not have HSE this will fail (based on 8MHz).

See https://github.com/stm32duino/Arduino_Core_STM32/blob/113f9e6368b5a093a4c71ba02bf8d820cd61f05e/variants/NUCLEO_F103RB/variant.cpp#L121

You can redefine at sketch level the clock config of your board:
#ifdef __cplusplus
extern "C" {
#endif

void SystemClock_Config(void)
{
 // Your clock config
}
#ifdef __cplusplus
}
#endif

fischermario, 2018-01-30:
Hi @fpistm!

Thank you very very much. The _SystemClock_Config_ function was the reason for all the trouble.

I have added

```
#ifdef __cplusplus
extern "C" {
#endif

void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

  /**Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = 16;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    while(1);
  }

  /**Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    while(1);
  }

  /**Configure the Systick interrupt time 
  */
  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);

  /**Configure the Systick 
  */
  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}

#ifdef __cplusplus
}
#endif
```

at the end of my sketch and it works great. I have also tried Firmata and it works too! For testing Firmata I used the firmata_test tool that was recommended in the [Wiki](https://github.com/stm32duino/wiki/wiki/Firmata-example).

I noticed that every time when I start the tool it shows me a different set of available pins (both digital and analog). For example it shows available analog pins (e.g. A0, A2, A4, A6, A8, A10, A11, A13) and I can successfully read their values. If I close the tool and start it again I get a different set of analog pins (e.g. A1, A3, A5, A7, A9, A11, A12). It seems completely random. Also the digital pins differ each time.

Is this intentional or just a bug?

fpistm, 2018-01-30:
It's a bug of the firmata test software.

fpistm, 2018-01-30:
Closed as it solved

closed 2018-01-30 by fpistm