stm32duino/Arduino_Core_STM32 #210: Core 2018.1.18 can not run in the Eclipce

Ronzheng18098, 2018-01-30:
Hello Frederic,

Teat board:   Nuleo STM32L432KC

(1) can you help us to find out why the stm32 core 2018.1.18  can not be run in the Eclipse IDE with "Eclipse c++ IDE Arduino 3.0" plugin . 

(2)and also can not run in the "The Arduino Eclipse IDE named Sloeber V4" plugin

i create a project named "Test",error info. as following:

Errors occurred during the build.
 Errors running builder 'org.eclipse.cdt.core.build.CBuilder' on project 'Test'.
 Undefined key build.source.path
 Undefined key build.source.path

it's ok for stm32 core2017.11.24 running with Eclipse  with the above mentioned plugin

Thanks Very much!!!!!

fpistm, 2018-01-30:
Unfortunately, we currently doesn't support another IDE than Arduino.
` build.source.path` is an Arduino properties which is the path to your sketch folder (where is the .ino) so probably plugins do not deal with it properly and should manage it. 
The previous version of the core does not use it that's why it's working.

closed 2018-01-30 by fpistm

Ronzheng18098, 2018-01-31:
thanks!!!! very much does it possible the next version support the Eclipse IDE, the same as the core stm32 2017.11.24?

fpistm, 2018-01-31:
The issue comes from the plugin.
It have to handle, properly Arduino property: `build.source.path`
Core need it so I will not remove it.

engrziarehman, 2020-02-10:
In new Eclipse C/C++ IDE, this Arduino plugin is built-in, and it is working absolutely fine with AVR and other Cores, but Eclipse is failed to build this core, please solve this 'build.source.path' problem, so that Eclipse or Sloeber could handle this core as they are handling other cores.

jandrassy, 2020-02-10:
in Sloeber you can build if you in platform.txt comment out
`recipe.hooks.prebuild.3.pattern`
and add SrcWrapper library manually

but the upload over bootloader doesn't work. I think because Sloeber sends a DTR reset before starting the upload tool. I am playing with this right now.

(the title of this issue refers to Roger Clark's core?)