stm32duino/Arduino_Core_STM32 #212: I2C slave failed

larsstenberg, 2018-01-30:
BOARD: NUCLEO-F091RC

Connected a NanoPi Neo Air (master) with the nucleo when running i2cset-y 0 0x09 on the NPI the nucleo hangs. Serial stops, blinking stop. It just freezes.

Code running on the nucleo:
`#include <Wire.h>

void setup()
{
  Wire.begin(0x15);                // join i2c bus with address #4
  Wire.onReceive(receiveEvent); // register event
  Wire.onRequest(requestEvent);
  Serial.begin(115200);           // start serial for output
}

void loop()
{
  delay(100);
}

void receiveEvent(int howMany)
{
  Serial.println(howMany);
  while(1 < Wire.available()) // loop through all but the last
  {
    int c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.print(c);         // print the character
}

void requestEvent()
{
  Wire.write("hello");
}`

1)
It seems to be a problem in the Wire.c when running as a slave the allocateRxBuffer never gets called. So first fix seems to be to add:

`  // copy twi rx buffer into local read buffer
  // this enables new reads to happen in parallel
+  if(rxBuffer == nullptr){
+    allocateRxBuffer(BUFFER_LENGTH);
+  }
  memcpy(rxBuffer, inBytes, numBytes);

and chaging allocateRxBuffer to static.

Now the freeze stopped.

2) It also seems to be an error with the numBytes received from master.
Changing line 541 in twi.c
from: nbData = I2C_TXRX_BUFFER_SIZE - obj->handle.XferCount;
to: nbData = I2C_TXRX_BUFFER_SIZE - obj->handle.XferSize;

Seems to correct the problem.

Am I doing something wrong or this a actual bug?

fpistm, 2018-01-30:
Hi @larsstenberg 
yes seems to be a bug :) thanks to report.

fpistm, 2018-02-02:
Hi @larsstenberg 
I've tested Master/receiver Wire example and it is now ok.
Could you review the patch, please?
Thanks in advance

commit 86e0bbcd (2018-02-02): Fix I2C slave issue: Rx buffer allocation

Fix #212
commit b1b12dec (2018-02-02): Fix I2C slave issue: return correct size

Fix #212
closed 2018-02-02 by fpistm