stm32duino/Arduino_Core_STM32 #215: NMEA2000 support

jiauka, 2018-01-31:
Hi and many thanks for your library.

I'm working on a library to support NMEA2000 can bus stuff
https://github.com/jiauka/NMEA2000_stm32f1

However, NMEA2000 implements a FASTPACKET and priorities that requires to be send in some order.

The best solution is to have a
HAL_StatusTypeDef HAL_CAN_Transmit(CAN_HandleTypeDef* hcan, uint32_t Timeout, uint32_t transmitmailbox)

and  a
HAL_StatusTypeDef HAL_CAN_Transmit_IT(CAN_HandleTypeDef* hcan, uint32_t transmitmailbox)

In such a way that the code can choose which TX mailbox to use.

This a common problem on many CAN bus libraries, it has been solved for teensy/mcp and others.

Can you make those mods?

Or as an alternative, I can  fork your code and modify it at least for the STM32F1xxx and STM32F3xx (the ones I have). If so, is there any chance to push those mods into the mainstream git?

thanks a lot in advance!

j.

fpistm, 2018-01-31:
Hi @jiauka 
Honestly, I will not have time to investigate this soon,  so if you can do a PR, this will be great. Any contribution are welcome ;)
Thanks in advance.

jiauka, 2018-01-31:
Thanks, I will do, at least for stm32f1xx

fpistm, 2018-10-12:
Hi @jiauka,
I close this issue as no update since a while. Any PR is welcome.

closed 2018-10-12 by fpistm